tttapa/Control-Surface #3: MIDI IN / LED with multiplexer ?

brian12112, 2018-03-21:
I know this library is under construction, but my questions is  there is a way or did it support using Multiplexer for MIDI Input / LEDs for the Final/Alpha Release?

Thank You

brian12112, 2018-03-22:
Wow, Its a great news. This is I be waiting for.  Thank you tttapa.

closed 2018-03-22 by brian12112

tttapa, 2018-03-23:
Note that you don't have to use RGB LEDs, you can use ordinary LEDs as well, and you can use any number of shift registers.  

For example, if you want a single shift register with 8 normal LEDs:
```cpp
 ShiftRegisterOut ShiftReg(dataPin, clockPin, latchPin, LSBFIRST, 8); 
  
 MIDI_LED leds[] = { 
   {ShiftReg.pin(0), MCU::MUTE_1, channel}, 
   {ShiftReg.pin(1), MCU::MUTE_2, channel}, 
   {ShiftReg.pin(2), MCU::MUTE_3, channel}, 
   {ShiftReg.pin(3), MCU::MUTE_4, channel}, 
   {ShiftReg.pin(4), MCU::MUTE_5, channel}, 
   {ShiftReg.pin(5), MCU::MUTE_6, channel}, 
   {ShiftReg.pin(6), MCU::MUTE_7, channel}, 
   {ShiftReg.pin(7), MCU::MUTE_8, channel},  
 }; 
```

brian12112, 2018-04-03:
Wow, I am consider to using it with Standalone LEDs. Thank You again tttapa. It helps me a lot!

brian12112, 2018-04-08:
Just one last questions. There is a solution to using multiplexer or shift register for the buttons? Thanks B4!

tttapa, 2018-04-08:
You can use `AnalogMultiplex` for buttons as well.
Or you can use a button matrix.

brian12112, 2018-04-09:
Thanks. I'll give it a try.