tttapa/Control-Surface #4: fix Issue noise mux founded!

userdsp, 2018-04-11:
https://www.youtube.com/watch?v=TTdOkhqLSr4
Please watch this will explane my issue noise issue 74hc40**
The cure is pull up resistor, but then all channels become with values over the level 0 when the pots are on minimal position.
Is there a way to change the value from 0-127 to x-127?

tttapa, 2018-04-12:
I don't speak Russian, so I have no idea what the video is about.  
Adding those resistors seems like a strange way to reduce noise.

Yes, you can change the value range using the [`AnalogBase::map()`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/classAnalogBase.html#ae97be3e099c45b1cbc199fc952ab5d06) and [`Bank::map()`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/classBank.html#a59d286595e6356c09da49eb4bfb0f8d8) functions.  
Take a look at [Ex.09.Bank-Calibrate](https://tttapa.github.io/Control-Surface/Doc/Doxygen/Ex_809_8Bank-Calibrate_8ino-example.html).  

Note, however, that adding a pull-down resistor to the potentiometer completely messes up the linearity:
[image]
[Potentiometer-pull-down.xlsx](https://github.com/tttapa/Control-Surface/files/1902359/Potentiometer-pull-down.xlsx)

This might be interesting: [http://forum.arduino.cc/index.php?topic=540051.0](http://forum.arduino.cc/index.php?topic=540051.0)

userdsp, 2018-04-12:
Lol, I was feeling there is something like this, but now when I saw this graph I understand it why. Thank you! I don't know why, but I have many brand new multiplexers, arduino uno and now teensy but both provide  the same issue. Maybe the breadboards make me those trouble with some lose connections :/

tttapa, 2018-04-12:
If you use pull-down resistors that are large enough, the influence could be neglegiable. But then the question is: does it help against the fluctuations ...

You can try different values for yourself in the spreadsheet I posted before.

userdsp, 2018-04-12:
Do you saw the pull resistor is connected to the common input pin. I'm not sure will he cmesses up the linearity.

userdsp, 2018-04-14:
Maybe you will like this :) 
Using an HDD Motor as a Rotary Encoder?!
https://www.youtube.com/watch?v=tjCJ3MlFt7g