tttapa/Control-Surface #5: possibility of motorized faders

oscarosky, 2018-04-27:
Could it include the reading of messages (MIDI_Input "PitchBend") to manipulate motorized faders?

tttapa, 2018-04-27:
Yes, but it requires specialized timing, and complicated control systems. That's why I didn't include it with the library (yet).

oscarosky, 2018-06-07:
I agree with you. 
The behavior of the system gets worse as new elements are added. Maybe the control of the faders needs a dedicated microcontroller.
We will also talk in a few weeks about the level meters on PWM.
A greeting.

masydoblig, 2018-06-10:
i was thinking that also if we put the faders on a dedicated controller 

here is the design i'm working on for my studio. It is for Cubase.
i the code works really well as a generic controller 

would love to get 8 motor faders working from one board so i can program them in banks of 8

[image]

masydoblig, 2019-10-20:
Any progress with the faders ???

tttapa, 2019-10-20:
I posted my code here: https://github.com/tttapa/MotorFader

I'm not satisfied with the results, and it's a lot of work to integrate it with the Control Surface library, and I don't have the time. I also have to add a Kalman filter and a better way to tune the PID controller dynamically.

tttapa, 2021-07-19:
For future reference:  
Motorized faders are now supported, you can find the documentation at https://tttapa.github.io/Pages/Arduino/Control-Theory/Motor-Fader and the code in [tttapa/Control-Surface-Motor-Fader](https://github.com/tttapa/Control-Surface-Motor-Fader).

closed 2021-07-19 by tttapa