tttapa/Control-Surface #6: Motorised Faders

masydoblig, 2018-05-15:
Looking through the files i can see that there are several ways to receive the incoming midi messages from Cubase 8.5 but i am using it a a generic remote. that way io can programm my own functions.
and it works great.
I have managed to get everything working except the shift registers :(  any help there with your schematics would be great. i think its my wiring that's wrong

But....

How can i modify this to give me the data from the MIDIcc message to control the motor fader controlled on a Hbridge for instance. https://github.com/codyhazelwood/motorized-fader-arduino such as this.

Other than that this code is awesom and works great on my 2560 with HIDUINO. NICE WORK

sorry im not very good a coding. i have been looking through all the LIB source code and i think i can do it in a similar way to your VU input for instance.

In cubase i have turned the transmit and receive button on and can see on the RX LED on the Board that every time i switch banks in cubase it is sending the data for the Fader position.

I just cannot work out how to use it  :(
[image]

#include <Control_Surface.h> // Include the library

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)
const uint8_t addresses[4][4] = {   // the note numbers corresponding to the buttons in the matrix
  {  1,  2,  3, 13 },
  {  4,  5,  6, 14 },
  {  7,  8,  9, 15 },
  { 10, 11, 12, 16 },
};

// Create a new instance of the class 'ButtonMatrix', called 'buttonmatrix', with dimensions 4 rows and 3 columns, with the rows connected to pins 2, 3, 4 and 5
// and the columns connected to pins 6, 7 and 8, that sends MIDI messages with the notes specified in 'addresses' on MIDI channel 1, with velocity 127
ButtonMatrix<4, 4> buttonmatrix( {14, 15, 16, 17}, {18, 19, 20, 21}, addresses, 1, velocity);

const uint8_t BANK_LEFT = 0x2E;
const uint8_t BANK_RIGHT = 0x2F;
const uint8_t LED1 = 61;// Note number 60 is defined as middle C in the MIDI specification
const uint8_t LED2 = 62;
const uint8_t LED3 = 63;
const uint8_t LED4 = 64;
const uint8_t LED5 = 65;
const uint8_t LED6 = 66;
const uint8_t LED7 = 67;
const uint8_t LED8 = 68;
const uint8_t LED9 = 69;
const uint8_t LED10 = 70;
const uint8_t LED11 = 71;
const uint8_t LED12 = 72;
const uint8_t LED13 = 73;
const uint8_t LED14 = 74;
const uint8_t LED15 = 75;
const uint8_t LED16 = 76;
const uint8_t test = 60;

//_____________________________________________________________________________________________________________________________________________________________________________________________

MIDI_LED leds[] = {
    {22, LED1, 1},
    {23, LED2, 1},
    {24, LED3, 1},
    {25, LED4, 1},
    {26, LED5, 1},
    {27, LED6, 1},
    {28, LED7, 1},
    {29, LED8, 1},
    {30, LED9, 1},
    {31, LED10, 1},
    {32, LED11, 1},
    {33, LED12, 1},
    {34, LED13, 1},
    {35, LED14, 1},
    {36, LED15, 1},
    {37, LED16, 1},
};

Digital switches[] = {
    {2, 0x10, 1, velocity}, 
    {3, 0x11, 1, velocity},
    {4, 0x12, 1, velocity},
    {5, 0x13, 1, velocity},
    {6, 0x14, 1, velocity},
    {7, 0x15, 1, velocity},
    {8, 0x16, 1, velocity},
};

Analog faders[] = {
    {A0, MIDI_CC::Channel_Volume, 1},
    {A1, MIDI_CC::Channel_Volume, 2},
    {A2, MIDI_CC::Channel_Volume, 3},
    {A3, MIDI_CC::Channel_Volume, 4},
};

Digital bankswitch[] = {
     {9, BANK_LEFT, 1},
     {10, BANK_RIGHT, 1},
};

//_____________________________________________________________________________________________________________________________________________________________________________________________

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  Control_Surface.refresh();
}

masydoblig, 2018-05-15:
I was just reading the other issues and noticed someone else requesting the motorised faders
It really would be awesom to get that on there
But seriosly this code works great in cubase as a generic remote
I am really impressed

masydoblig, 2018-05-19:
so everything seems to be working,,,now all we need is some amazing tttapa code for the fader motors.

so if i have this correct we will need to block the signal coming from the fader while the fader is getting moved by the daw. and make the code so the fader only sends a signal when the cap sense is active, ie the daw has priority over the fader movement,,,,and only when cap sense is active we need to give the fader send signal priority over the DAW fader control signal.

ps this code is still awesome

tttapa, 2018-05-20:
I don't have time to look into it at the moment (finals), I'll get back to you in early July.

masydoblig, 2019-02-10:
Hey Dude 

So i have managed to complete the the circuit boards for the buttons and encoders and put it all together 
it all seems to be working well, how are you going with the touch sense motorised fader code?

i have decided to make them all separate units ie fader module, EQ module and master module 

is there any way to daisy chain the boards together?
so i can have one controller in Cubase to program 
 
[image]

tttapa, 2019-02-10:
I've worked on it for the last week or so, and I got them more or less working. I decided to offload the controllers to separate ATtiny85 chips. Running nine motorized faders that update at 1 kHz, while updating all other stuff as well is just too complicated, and gives problems with timing.
Setting up the PWM timers for the motors, timer interrupts for the controllers, and triggers for the ADC to run "in parallel" with the other code is just not happening. If each fader has its own controller, it's much easier.

It's not the optimal solution, but I don't think I can do much better at the moment.

The code is in a private repo right now, but I'll clean it up and make it public ASAP.

The one thing I still have to do is add the touch support. The problem is that the ATtiny85 has only 5 IO pins. You need 2 to control the motor, 2 for I2C communication with the main Control Surface Arduino/Teensy, one for reading the analog input, and one for reading the touch sensor.

There is a sixth IO pin, but it's used as the reset pin. That means that you can use it, as long as the voltage doesn't go below 2.5V or so, otherwise, the ATTiny resets.
Alternatively, you can use a high voltage programmer to disable reset on that pin, and use it as a more or less full IO pin, but that's not exactly user friendly.

Any thoughts on this?

***

I'm going to need more information about daisy chaning the boards together. What microcontroller are you using in each of the strips? Do you have any pins free for communication, like UART or I2C?

If so, you could use the ExtIO functions (with some additions) to create one master controller that sends out IO commands to the other slave controllers (e.g. "slave 1, turn on LED 5", "slave 2, give me the state of button 2"). 

Reading rotary encoders might require some modifications, but nothing too hard.

Your picture looks great, by the way :D

masydoblig, 2019-02-11:
Hi Peter 

Thanks for the reply 

I have been thinking about it all day (yay Motorised Faders)
We should have a face time meeting or a skype to chat about all this 

It seems you are already using I2C (or similar) to the fader chips, so why not just make each fader controlled from a arduino Nano that should solve the pins issue (and the are cheap to get off ebay).
 Can you send me a block diagram of how you have it set up so i can get a better picture.

I think its a great idea to make each fader with its own controller, my original plan was to just get a 16 port usb hub and make 16 individual controllers like the pic above (8 fader strips and 8 eq strips).
but it makes more sense to link them all together with I2C or similar to a master controller using all 16 Midi channels. I just have no idea how to do it. LOL
[image]

If you were to go down that road it would be pretty easy to design a PCB with the controllers and Hbridge all in one for the fader Module

As im sure you are aware the touch support is really important as it allows the fader motor to be cut out when manually moving the fader (to avoid burning out the motor etc).

I am definitely not good at programming these things, it took me a while to work out the existing setup
using your code (which is awesome by the way)
but i kept it simple.
I am using an Arduino Mega so i don't have to use the shift registers etc. (i was struggling with the code to get it to work)
each LED has its own pin and each button has its own pin.
i got some pcbs made through easy eda for all the buttons and LEDS 
it seems to work really well and its very fast scrolling through channels 

[image]

Will Your code be able to drive one fader with touch sense from my board? (with out the update issues you mentioned above) as there are only 
9 buttons (including button on encoder top)
8 led
1 encoder
1 fader

Here Is the code i have in the boards at the moment (i havnt been using the MCU parts so i can just make it a generic controller )

#include <Control_Surface.h> // Include the library

const uint8_t Controller = 0x14; // MIDI Control Change controller number
const int speedMultiply = 2;     // No change in speed of the encoder (number of steps is multiplied by 1)

RotaryEncoder encoder(2, 3, Controller, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);// comes through as a controller on address 20
RotaryEncoder encoder2(4, 5, Controller, 2, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);// other digital pins work

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)

const uint8_t LED1 = 40;// Note number 60 is defined as middle C in the MIDI specification
const uint8_t LED2 = 41;
const uint8_t LED3 = 42;
const uint8_t LED4 = 43;
const uint8_t LED5 = 44;
const uint8_t LED6 = 45;
const uint8_t LED7 = 46;
const uint8_t LED8 = 47;
const uint8_t LED9 = 48;
const uint8_t LED10 = 49;
const uint8_t LED11 = 50;
const uint8_t LED12 = 51;
const uint8_t LED13 = 52;
const uint8_t LED14 = 53;

const uint8_t channel = 1;

//_____________________________________________________________________________________________________________________________________________________________________________________________

MIDI_LED leds[] = {
    {40, LED1, 1},
    {41, LED2, 1},
    {42, LED3, 1},
    {43, LED4, 1},
    {44, LED5, 1},
    {45, LED6, 1},
    {46, LED7, 1},
    {47, LED8, 1},
    {48, LED9, 1},
    {49, LED10, 1},
    {50, LED11, 1},
    {51, LED12, 1},
    {52, LED13, 1},
    {53, LED14, 1},
};

Digital switches[] = {
    {22, 0x00, 1, velocity},
    {23, 0x01, 1, velocity},
    {24, 0x02, 1, velocity},
    {25, 0x03, 1, velocity},
    {26, 0x04, 1, velocity},
    {27, 0x05, 1, velocity},
    {28, 0x06, 1, velocity},
    {29, 0x07, 1, velocity},//comes in on channel 1 address 22
    {30, 0x08, 1, velocity},
    {31, 0x09, 1, velocity},
    {32, 0x10, 1, velocity},
    {33, 0x11, 1, velocity},
    {34, 0x12, 1, velocity},
    {35, 0x13, 1, velocity},
    {36, 0x14, 1, velocity},

    
};

Analog faders[] = {
    {A0, MIDI_CC::Channel_Volume, 1},// comes through as address 7
};

//_____________________________________________________________________________________________________________________________________________________________________________________________

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  Control_Surface.refresh();
}

tttapa, 2019-02-11:
> It seems you are already using I2C (or similar) to the fader chips, so why not just make each fader controlled from a arduino Nano that should solve the pins issue (and the are cheap to get off ebay).

That's a great idea, I didn't know they were so cheap. With some clever coding, I could even get up to 6 motorized faders per Nano. (There are 6 PWM channels.)

One thing I have to check is the PWM frequency: The PWM frequency I'm using right now is 31.25 kHz, using an 8 MHz clock in fast PWM mode. The Arduino Nano has a 16 MHz clock, so the PWM frequency would be 62.5 kHz, which is too high (the H bridges/motor drivers cannot handle this). The next possible frequency is 7.813 kHz, which is in the audible range.  
A solution would be to use phase-correct PWM, it's about two times slower, so we could use 31.25 kHz again. The problem in this case is the triggering of the ADC, which happens in the middle of the high/low part of the PWM pulse, instead of the switching time.  
I don't know if this will cause more or less electromagnetic interference in the position and touch measurements.

> Can you send me a block diagram of how you have it set up so i can get a better picture.

The picture you posted is pretty close:   
I have one master Arduino, that communicates with the ATtiny over I2C. The ATtiny is connected to the motor controller and the potentiometer of the motor fader.  
The master just sends commands like: "set fader target position to _x_", or "request fader position+touch". The ATtiny runs the controller, and tries to get the fader to the target position. When the master requests the position, the ATtiny responds with the current position of the fader, and whether it is touched or not.  
It also turns off the motor automatically when the fader is touched.

This is the code I'm using on the ATtiny (touch code is commented out for now, because of a lack of pins):  
https://github.com/tttapa/MotorFader/blob/master/Arduino/ATtiny85/ATtiny85-I2C/ATtiny85-I2C.ino  
And this is the code on the Master Arduino:  
https://github.com/tttapa/MotorFader/blob/master/Arduino/I2C-master/I2Cmaster/I2Cmaster.ino  
It just sets the target position from 25% to 75% and back again, every 2 seconds.  
It's also possible to set the controller tuning parameters over I2C.

> Will Your code be able to drive one fader with touch sense from my board?

Most likely, yes.  
The Arduino MEGA has many timers you could use for the fader controllers, and since there's nothing else using the ADC, adding the controller code should be trivial (very similar to the ATtiny85 code).

If you have normal potentiometers, or multiple motor faders on a single Arduino, it's a bit harder.  
To keep the library general (and simple), and to support as many configurations as possible, it makes sense to use an external controller for the motor faders, but I don't think it's strictly necessary in your case.  
The problem is that I don't have an Arduino MEGA to experiment with.

masydoblig, 2019-02-11:
well if you can help me with some code for the fader and touch sense for me to try with my mega and L298 H bridge i will test it out. (like i said im crap at coding)
And see what kinds of reactions i get.

Just tell me which pins to connect to 
below is the h bridge i have 

[image]

I really think the Nano is the go then if you can get the PWM Frequency sorted 
and would be very easy to duplicate onto a custom PCB with h bridge included 

I thought i had a spare Nano Lying around but i only have the one im using for my programmer board for the mega (works awesome i just put it on the mega every time i need to flash it)
[image]

i also have a spare due and uno that could be used to test the master slave setup

masydoblig, 2019-02-11:
This could also be a good thing as the controller for each module

[Image]

https://github.com/tttapa/MotorFader/blob/master/Arduino/ATtiny85/ATtiny85-I2C/ATtiny85-I2C.ino
https://github.com/tttapa/MotorFader/blob/master/Arduino/I2C-master/I2Cmaster/I2Cmaster.ino

masydoblig, 2020-10-15:
https://github.com/saarts/MIDI/blob/master/MIDI_flying_fader/MIDI_flying_fader.ino

this guy has some interesting code for the motorised faders support for Arduino @tttapa 
uses the midiusb.h library
very responsive in the you tube video 

Let me know what you think

tttapa, 2020-10-17:
The responsiveness is not really an issue, the main problems I have are:

- It is very hard to tune the faders to track the setpoint accurately.
- Recompiling+uploading the sketch to change the tuning is cumbersome, so I'd need an easy to use interface to send tuning parameters to the Arduino and the ability to store them in EEPROM or other NVS.
- The timing of the control loop is hard to get right in a platform-independent way, and should ideally be run in a Timer interrupt context. Reading the ADC in an ISR context is a nightmare as well, because you have to share a single ADC hardware peripheral between the control loop in interrupts and the `analogRead` calls in the main program.
- Solving these issues isn't impossible, but they would require a lot of time to get right, and a lot of code for the different supported architectures. While I'm familiar with the AVR Timers and ADC architecture, studying hundreds of pages of ARM datasheets is not something that I have time for right now.
- Documenting how to get motorized faders working (including hardware, tips for tuning, etc.) would take a lot of time, time I should probably be spending on my thesis instead.

The main conclusion is that motorized faders are too platform-specific, too time-consuming, and not user-friendly enough to be added as a part of the library.  
Control Surface does provide tools to receive and send MIDI data, so you could implement motorized faders yourself for your specific platform, but it does require some basic control theory knowledge, and you have to be familiar with the timers, interrupts and ADC of the chip you're working with.

tttapa, 2021-07-19:
For future reference:  
Motorized faders are now supported, you can find the documentation at https://tttapa.github.io/Pages/Arduino/Control-Theory/Motor-Fader and the code in [tttapa/Control-Surface-Motor-Fader](https://github.com/tttapa/Control-Surface-Motor-Fader).

closed 2021-07-19 by tttapa