tttapa/Control-Surface #10: add led on/off while midi send note

pandeagus, 2019-01-27:
### Description of the problem or question
i make usb midi foot controller from your library and your library very good and easy to use .. but i use 8 button and 8 led... i want when button press
led is on and when button press again led is off.. can you teach me how can make easy away for the code or method modif your library?
this is my code 

### Full code 

```cpp
#include <MIDI_Controller.h>

int led1 = 10;
int led2 = 16;
int led3 = 14;
int led4 = 15;
int led5 = 18;
int led6 = 19;
int led7 = 20;
int led8 = 21;

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)
const uint8_t C4 = 60;
const uint8_t D4 = 63;
const uint8_t E4 = 64;
const uint8_t F4 = 65;
const uint8_t G4 = 67;
const uint8_t C3 = 48;
const uint8_t B4 = 71;
const uint8_t D3 = 50;

Digital button1(2, C4, 1, velocity);
Digital button2(3, D4, 1, velocity);
Digital button3(4, E4, 1, velocity);
Digital button4(5, F4, 1, velocity);
Digital button5(6, G4, 1, velocity);
Digital button6(7, A3, 1, velocity);
Digital button7(8, B4, 1, velocity);
Digital button8(9, 72, 1, velocity);

void setup () {}

void loop() {
  MIDI_Controller.refresh();
}
```

pandeagus, 2019-01-28:
[image]

i get error when getting start using 8 button.. i just want make usb midi with 8 button can MIDI CC - MIDI NOTE together and 8 potentiometer and 8 led indicator switch.. 

im sorry bother you...
thanks

tttapa, 2019-01-28:
`Digital` has been replaced by `NoteButton`. You cannot use the old MIDI Controller classe, they have been replaced by improved versions with a different name.

The documentation for all of these classes can be found here: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d7/dcd/group__MIDIOutputElements.html

tttapa, 2019-01-28:
If you really want to send both notes and CC at the same time (I still don't understand why it's necessary), you can instantiate a `NoteButton` and a `CCButton` on the same pin. 

Alternatively, use this with the latest version of the library:
```cpp
// Include the Control Surface library
#include <Control_Surface.h>

class DigitalNoteAndCCSender {
  public:
    static void sendOn(MIDICNChannelAddress address) {
      Control_Surface.MIDI().sendNoteOn(address, velocity);
      Control_Surface.MIDI().sendCC(address, onValue);

    }
    static void sendOff(MIDICNChannelAddress address) {
      Control_Surface.MIDI().sendNoteOff(address, velocity);
      Control_Surface.MIDI().sendCC(address, offValue);
    }
    constexpr static uint8_t velocity = 0x7F;
    constexpr static uint8_t onValue = 127;
    constexpr static uint8_t offValue = 0;
};

using NoteAndCCButton = MIDIButton<DigitalNoteAndCCSender::sendOn, 
                                   DigitalNoteAndCCSender::sendOff>;

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

using namespace MIDI_Notes;

// Instantiate the momentary push buttons that send out 
// MIDI Note and CC events.
NoteAndCCButton buttons[] = {
  {2, note(C, 4)}, // digital input pin, note number
  {3, note(Db, 4)},

  // If you want to set the channel specifically, you can use:
  // {2, {note(C, 4), CHANNEL_1}},
};

void setup() {
  // Initialize everything
  Control_Surface.begin();
}

void loop() {
  // Update the control surface
  Control_Surface.loop();
}
```

pandeagus, 2019-01-29:
I am very lucky because you helped me in working on this project. and for that I thank you. and you are the best teacher in teaching me about midi controllers. and about MIDI CC and MIDI NOTE it's complete thanks to all your help. maybe later I will ask you more about the midi controller and hope you don't get bored by answering my questions.

tttapa, 2019-01-29:
No worries. Glad to hear you got it working!

closed 2019-01-29 by tttapa

guillecip, 2019-03-20:
In the example of button with led once you push the button the led is always on. 
Can we turn off the led in a toggle funcion? For example in Traktor if a turn on a FX a light in the screen turn on. When I turn off the FX the screen turn off. Can I do these with an external LED?

masydoblig, 2019-03-20:
Like this one for cubase?
YouTube link below

https://youtu.be/sIzhsGYoD2U

guillecip, 2019-03-21:
Yes, just like that.

masydoblig, 2019-03-21:
Yes you can. Petes control surface library is pretty amazing.
But to get the LEDs to come on when you activate a function  the DAW needs to output a midi signal to your controller you have built.
I have no experience with tractor but if it has mackie control it is completely possible.

There is a full guide to this library and a link to it from the repository start page (In the description section at the bottom)

There is also a non midi feedback button LED constructor in the examples I'm pretty sure.