tttapa/Control-Surface #11: HOW TO Modif LIBRARY

pandeagus, 2019-02-02:
HI PIETER..
I want ask about library control surface
how i can make CCButtonlatched.hpp  send MIDI CC from modif NoteButtonLatched.hpp.
i want make like example like pushbutton no midi feedback .. but with MIDI CC send.. 
thanks
Best Regards.

pandeagus, 2019-02-13:
Hi pieter its me again.. 
Make it possible if this library can use eeprom to store midi cc or midi note setting
Maybe setting can store in ic at24c02 or else.. 
Im sorry i annoying again..
Once question again.. 
Can midi and usb midi run together using pro micro atmega32u4 ?
 Thanks.. 
Best Regard

tttapa, 2019-02-13:
> Make it possible if this library can use eeprom to store midi cc or midi note setting

Could you elaborate? Please post your full code, and clearly indicate what you want to save.

> Maybe setting can store in ic at24c02 or else..

Why? Almost all Arduinos have EEPROM built-in, there's no need for external memory.

> Can midi and usb midi run together using pro micro atmega32u4 ?

Yes, but the Control Surface class can't receive from two MIDI inputs simultaneously.

pandeagus, 2019-02-15:
hi pieter I mean it's like meloaudio midicomander where
every pushbutton can we set the midi cc number as needed and save it on eeprom.

Sorry, I haven't shown the code because the LCD and other materials have not arrived from the shipment, so I can't try it yet.

maybe this is the midi comannder riveiw link
https://youtu.be/LPVNMxxGk0U

thanks pieter..
im sorry for the trouble i made

tttapa, 2019-02-16:
You can already set the CC numbers by just uploading a new sketch. If you want to be able to change them at runtime, you certainly could, but you would have to write a lot of supporting code for the display, the menus, the interface, the buttons to control it, etc.  
The library cannot help you with that part.

That being said, the library does supports assigning multiple addresses to one MIDI element, and then allows you to select which of these addresses to use.  
Have a look at the ManyAddresses namespace: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d6/d79/namespaceManyAddresses.html>  
It's just a proof of concept at the moment, but adding CCButtons and NoteButtons etc. is trivial, and I'll be adding them when I find the time.

I'll also add getters and setters for the addresses to the MIDI elements classes, so you can change the addresses at run time. This does come with some caveats though: You have to make sure that you don't change the addresses when the pushbutton is pressed, as that would result in notes that keep on playing for eternity.

Note that you can get and set the settings of the [selectors](https://tttapa.github.io/Control-Surface/Doc/Doxygen/d1/d54/classSelector.html), so you can save them to EEPROM if you want.

closed 2021-02-10 by tttapa