tttapa/Control-Surface #12: Can I use mux for digital read of buttons?

maykefreitas, 2019-02-12:


tttapa, 2019-02-12:
Yes.

closed 2019-02-12 by tttapa

maykefreitas, 2019-02-12:
@tttapa can you please give me an example of row to use `CD74HC4067` MUX with `CCButton`?

tttapa, 2019-02-12:
Just like this example: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/dd/dcc/md_Getting-Started.html>  
Then just pass the pin number to the CCButton contructor:
```cpp
CCButton button = {mux.pin(0), address};
```

maykefreitas, 2019-02-13:
Thank you @tttapa, congratulations for the library. It's awesome. I turned a broken POD HD 500 into a MIDI controller with the Control Surface library. Just need to make expression pedal works, it's using an IR transmissor and receptor.

tttapa, 2019-02-13:
Do you need help with the expression pedal? Do you have a picture, drawing or schematic?