tttapa/Control-Surface #13: rotary encoders

masydoblig, 2019-02-14:
Hi Peter 

How you going 

So i have updated my library to your latest version of control surface 
and just going through and changing all the namespaces 
but i cant find any examples of the rotary encoder code lines 

can you help me out or point me in the right direction where i can find it 

i was using the old RotaryEncoder for my pan control and now it throws an error on compiling 
says it does not name a type 

sorry to bother you with something so trivial (i really am terrible at any form of coding)

and thanks again you have put a lot of time and effort into this code and i really appreciate it

tttapa, 2019-02-14:
Hi,   
The `RotaryEncoder` class has been renamed to `CCRotaryEncoder`.  
The interface has been changed as well, you can find the documentation for it here:  
<https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d62/classCCRotaryEncoder.html>

My apologies for the lack of proper examples or better documentation, the library has gotten really huge, and I have to find ways to combine working on it with my studies.

masydoblig, 2019-02-14:
Dont apologize
Your work is awesome
Hopefully  I can get it sorted

What country are you in?
So I'm not interrupting your sleep from here in Australia

https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d62/classCCRotaryEncoder.html

tttapa, 2019-02-14:
I'm in Belgium, it's 11:11 am here, no worries :)

masydoblig, 2019-02-16:
i have tried everything to get the rotary encoder working with no luck 

either i have the incorrect encoder.H library or i am writing the code incorrect to create a new instance of CCRotaryEncoder

any help with an example would be greatly appreciated  

also in the MCU protocol what midi message is it sending out for the VU meters 

[image]

im hoping i can just send the midi data from cubase (which i can set the type of midi message on) to the controller and get the vu on a shift register working

tttapa, 2019-02-16:
I added a rotary encoder example: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d15/CCRotaryEncoder_8ino-example.html>

> also in the MCU protocol what midi message is it sending out for the VU meters

It uses MIDI Channel Pressure events: the high nibble of the data byte is the track (0-7) and the low nibble is the value (0x0-0xC) or clear overload (0xF) or set overload (0xE). 0xD has no meaning.

masydoblig, 2019-02-16:
ok looking at the code you have put up it i would have to use the mcu mapping 
is there any way for me to set specific channels and addresses like we could before?
so i can use the generic mapping in cubase 

RotaryEncoder encoder(2, 3, 0x14, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);// comes through as a controller on channel 1,address 20

or has this all changed?

i.e.

CCRotaryEncoder enc ={
        {2, 3},
        MIDI_CC::Pan, CHANNEL_1,
        1,
        4,
};

The Reason for me to do this is i can do alot more through the generic program function in cubase 
even create banks and flip modes

masydoblig, 2019-02-16:
this code below works
 
CCRotaryEncoder enc ={
{2, 3},
MIDI_CC::Pan,
1,
4,
};

it functions well and comes though to cubase on Channel 1 address 10 

but i cant change channel number

tttapa, 2019-02-16:
You have to add curly braces, the address is a single parameter:
```cpp
CCRotaryEncoder enc ={
  {2, 3},
  {MIDI_CC::Pan, CHANNEL_1},
  1,
  4,
};
```

masydoblig, 2019-02-16:
Using the generic controller function is also why i was asking about the vu meters 

when i set it to a Mackie controller in cubase my Vu meter works 
but i cant set a specific channel and address in the code to use with the generic controller

masydoblig, 2019-02-16:
[image]

masydoblig, 2019-02-16:
Ok that makes sense
it compiles now 

you are an awesome dude 

How did you go with finding some Arduino Nano Boards to test out your faders????

tttapa, 2019-02-16:
> when i set it to a Mackie controller in cubase my Vu meter works
but i cant set a specific channel and address in the code to use with the generic controller

Could you check what kinds of messages Cubase sends? The easiest way is to use a MIDI monitor on you computer, alternatively, enable debugging information in Settings.hpp: uncomment lines [17](https://tttapa.github.io/Control-Surface/Doc/Doxygen/db/d02/Settings_8hpp_source.html#l00017) and 19.

> How did you go with finding some Arduino Nano Boards to test out your faders????

I started working on it, but I don't have the right hardware here right now.

masydoblig, 2019-02-16:
Can you suggest a MIDI monitor to try 

i have midi ox installed but it wont get anything from cubase 

Feel free to Face time me on masydoblig@hotmail.com so you can see it all

tttapa, 2019-02-16:
I use MIDI Snoop on Linux, but any monitor that runs on your platform should work.

masydoblig, 2019-02-16:
hmmmmm i tried the debug but it wouldnt compile and just had errors

and for some reason the midi monitor wont connect when cubase is running

tttapa, 2019-02-16:
Did you pull the latest version of the library?

masydoblig, 2019-02-16:
yep yesterday

tttapa, 2019-02-16:
I pushed another change this morning, that should have fixed the debug issues.

Did you install the library using git? <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d4/d6d/md_Installation.html>  
In that case, you can just do `git pull` to update to the latest version.

masydoblig, 2019-02-16:
i have just downloaded the new version 
it compiles
and i have put the USBDebugMIDI_Interface into the code but it doesnt seem to do anything 
still works as a controller though 

i might have to give up for the night,
have good day pete

Here is what i have so far 

#include <Encoder.h>
#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

USBDebugMIDI_Interface midiInterface(115200);

const pin_t dataPin  = 12;  // Pin connected to DS of 74HC595
const pin_t clockPin = 13;  // Pin connected to SH_CP of 74HC595
const pin_t latchPin = 10;  // Pin connected to ST_CP of 74HC595

// Create a new shift register output connected to pins 12, 13 and 10,
// shift the data out with the most significant bit first.
// There are 8 outputs in total.
ShiftRegisterOut<12> sr(dataPin, clockPin, latchPin, MSBFIRST);

Channel channel = CHANNEL_1;
// Note: a VU meter of 12 LEDs will give the best effect, because
// in the MCU protocol, there are 12 values
MCU::VULEDs<12> vu = { 
  {0x01, channel}, 
  {{
    sr.pin(0),
    sr.pin(1),
    sr.pin(2),
    sr.pin(3),
    sr.pin(4),
    sr.pin(5),
    sr.pin(6),
    sr.pin(7),
    sr.pin(8),
    sr.pin(9),
    sr.pin(10),
    sr.pin(11),
  }},
  150, // decay time in milliseconds or MCU::VU::NO_DECAY
};

// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer(A0, {MIDI_CC::Channel_Volume, CHANNEL_1});

CCRotaryEncoder enc = {
    {2, 3},
    {MIDI_CC::Pan, CHANNEL_1},
    1,
    4,
};

using namespace MIDI_Notes;

// Instantiate the momentary push buttons that send out MIDI Note events.
NoteButton buttons[] = {
    {22, {note(C, -1), CHANNEL_1}},
    {23, {note(Db, -1), CHANNEL_1}},
    {24, {note(D, -1), CHANNEL_1}}, 
    {25, {note(Eb, -1), CHANNEL_1}},
    {26, {note(E, -1), CHANNEL_1}}, 
    {27, {note(F, -1), CHANNEL_1}},
    {28, {note(Gb, -1), CHANNEL_1}},
    {29, {note(G, -1), CHANNEL_1}},
    {30, {note(Ab, -1), CHANNEL_1}},
    {31, {note(A, -1), CHANNEL_1}},
    {32, {note(Bb, -1), CHANNEL_1}},
    {33, {note(B, -1), CHANNEL_1}},
    {34, {note(C, 0), CHANNEL_1}},
    {35, {note(Db, 0), CHANNEL_1}},
    {36, {note(D, 0), CHANNEL_1}},  
    // If you want to set the channel specifically, you can use:
    // {2, {note(C, 4), CHANNEL_1}},
};

MIDINoteLED leds[] = {
    {40, {note(E, 2), CHANNEL_1}},
    {41, {note(F, 2), CHANNEL_1}},
    {42, {note(Gb, 2), CHANNEL_1}},
    {43, {note(G, 2), CHANNEL_1}}, 
    {44, {note(Ab, 2), CHANNEL_1}},
    {45, {note(A, 2), CHANNEL_1}},
    {46, {note(Bb, 2), CHANNEL_1}},
    {47, {note(B, 2), CHANNEL_1}},
    {48, {note(C, 3), CHANNEL_1}},
    {49, {note(Db, 3), CHANNEL_1}},
    {50, {note(D, 3), CHANNEL_1}},
    {51, {note(Eb, 3), CHANNEL_1}}, 
    {52, {note(E, 3), CHANNEL_1}},
    {53, {note(F, 3), CHANNEL_1}},  
};

void setup() {
    // Initialize everything
    Control_Surface.begin();
}

void loop() {
    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
}

tttapa, 2019-02-16:
Keep the USBMIDI_Interface, don't use the debug interface, just enable debugging in Settings.hpp, see the link in a previous reply: https://github.com/tttapa/Control-Surface/issues/13#issuecomment-464343274

masydoblig, 2019-02-17:
all good close this issue if you want as the encoder is working well
i will start a new one for debugging

closed 2019-02-17 by tttapa