tttapa/Control-Surface #14: Debugging

masydoblig, 2019-02-17:
Hey there Pete

I have uncommented the lines 17 and 19 
can you please give detailed description of how to use debugging 

i cannot work out where to see the messages (i.e the arduino serial monitor)
and what must be connected 

then i can see what messages are being sent to the arduino for fader movements and vu signals 
to help with the programing of motor faders and vu meters 

For some reason i cannot get any of the software midi monitors to work with cubase 
it must take full control of the ports 

Cheers for all the help 
Sam

tttapa, 2019-02-17:
I think I know why it doesn't work. Are you using an Arduino Mega to test this?  
The Mega only has a single USB connection that operates as either MIDI over USB or Serial.  
On boards with native USB support (Teensy, Leonardo, Due, Zero, etc.) you can use MIDI and Serial at the same time.  
So on these boards, you can debug while doing MIDI.

On the Mega, it's a bit harder: you'll have to use a different serial port for debugging.

Do you have a USB to Serial adapter? If not, you can just use an Arduino Nano or something like that if you have one lying around. Upload an empty sketch to the Nano:
```cpp
void setup() {}
void loop() {}
```
And connect it to UART1 of the Mega: TX to TX1 and RX to RX1.

Then, in Settings.hpp, change the `DEBUG_OUT` definition to `Serial1`. After uploading the right code to the Mega, select the Nano again, and open the serial monitor at 115200 baud. You should be able to see the debugging information.

The principle of debugging is pretty straightforward: if you define `DEBUG_OUT`, debugging is enabled, and the library will write debugging information to the specified output. It will also call `DEBUG_OUTPUT.begin(115200)`.

The default debug port is `Serial`, but you cannot use that one on an Arduino UNO or Mega, because all MIDI messages also go over that Serial port.

masydoblig, 2019-02-17:
CHAMPION

Got it to work and worked out that the vu meter from cubase needs to be set to "aftertouch" on channel 1 with a max value of 12 or 24 (seems to be a slight calibration error when using cubase, i tried to change the floor correction in the mculed file but it didnt help much)
but atleast the vu meter works now 

The debbugging works well from the Mega to the uno 

I also checked what cubase send back to the controller when moving the fader on the screen 
it sends the same message as controller sends "channel 1 address 7 with 124 values 
if this helps for the motorised faders 

Any help with the calibration for the VU meters would be appreciated
if i set a max value to 12 (from cubase) only the first 6 LEDs light up
if i set a max value of 24 they all light up and the max is correct but the lower leds are lighting up even when the meter on cubase is low volume 

PETE IS THE BEST GUY IN BELGIUM

LOW VOLUME
[image]

HIGH VOLUME
[image]

tttapa, 2019-02-17:
Could you post the exact MIDI messages that are sent? E.g. `>>> D0 0C` or `>>> B0 07 7F`

masydoblig, 2019-02-17:
this is with the max value set to 24 goes from d0 0c (meter max) down to d0 00 (meter off)

>>> d0 0c
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 0b
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 0a
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 09
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 08
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 07
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 06
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 05
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 04
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 03
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 02
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 01
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 00
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated

masydoblig, 2019-02-17:
This is with max value set to 12 from d0 06 to d0 00

CN = 0
>>> d0 06
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 05
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 04
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 03
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 02
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 01
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated
CN = 0
>>> d0 00
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 101]:	Updating Channel Pressure elements with new MIDI message.
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 40]:	
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 44]:	MIDI message matches
[bool MIDIInputElement::updateWith(const MIDI_message_matcher&) @ line 47]:	Updated

tttapa, 2019-02-17:
Where did you select the "maximum value"? What is it? Could you send a screenshot?

The correct maximum _MIDI_ value is 12, (which is 0xC), so it's strange that it only goes up to 0xC when you select 24 as a maximum value.

masydoblig, 2019-02-17:
This is what cubase sends out when i move the fader(with the mouse on the screen) from 0 to max

CN = 0
>>> b0 07 02
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 03
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 04
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 05
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 06
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 07
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 08
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 0a
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 0b
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 0c
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 0d
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 0e
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 0f
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 10
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 11
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 12
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 14
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 15
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 16
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 18
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 19
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 1b
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 1d
[... 162 lines cut ...]
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 76
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 77
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 78
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 79
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 7b
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 7d
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 7e
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.
CN = 0
>>> b0 07 7f
[virtual void Control_Surface_::onChannelMessage(MIDI_Interface&) @ line 90]:	Updating CC elements with new MIDI message.

masydoblig, 2019-02-17:
yeah thats what i thought 

[image]

[image]

tttapa, 2019-02-17:
I'll try to install a demo version of Cubase, I'm not familiar with its MIDI settings.

masydoblig, 2019-02-17:
it works really well set at 24 but the lower LEDs of the meter are illuminated early 
almost like the cubase meter is on a log taper and the meter is on a linear taper 

the generic controller setup in cubase is a very deep dark secret of the DAW there is no information and no help from steinberg 
but i am finding it to be very powerful you can change anything in cubase (almost)
and any insert i add to a track comes up with all the parameters in the midi menu to manipulate 
you can also create custom banks and set one of the controller buttons to go straight to it (for controlling a specific VST insert)
hence my reasoning to not use the MCU protocol

that and its very fast

tttapa, 2019-02-17:
> almost like the cubase meter is on a log taper and the meter is on a linear taper

I suspect that it's a bug in Cubase. It's not possible to convert it to a log curve on the Arduino, because of the limited resolution [0,12] sent by Cubase, so it has to be done on the computer.

I'm currently working on a CC VU meter, that has higher resolution, and that could do the conversion on the Arduino.

masydoblig, 2019-02-17:
It's all good I'm just glad I got it mostly working
With your help :)
The cc vu could be a good thing. A 30 LED strip would look awesome bouncing away

Can you send me a message  when you get the motor faders working. I can't wait to get that sorted and working of the Mega.
CCmotorfader here we come

I really  can't thank you enough

closed 2021-02-10 by tttapa