tttapa/Control-Surface #15: Track Names for OLED

masydoblig, 2019-02-19:
Hi Pete 

More of a question than an issue 

I have been looking through the getting stated doc for the oled displays and i am wondering if there is a "Track name" display for the oled display 
so i can see what track the controller is on 

i set cubase as a mackie controller and changed the track name- in the debugging serial monitor it says Data bye ignored when it send the message through to the controller 

So i think it is sending out the messages 

What are your thoughts

Cheers Sam

tttapa, 2019-02-19:
You can try commenting out the `IGNORE_SYSEX` macro in Settings.hpp, and see what it's sending.

masydoblig, 2019-02-20:
Yep its sending the track names as per the character map

MACKIE HUI SMALL DISPLAY CHARACTER SET
--------------------------------------

Reverse engineered by AgeMan, 2010.

Hex   Display   Hex   Display   Hex   Display   Hex   Display

00    ì         10    è         20    Space     30    0
01    ↑         11    Æ         21    !         31    1
02    →         12    æ         22    "         32    2
03    ↓         13    Å         23    #         33    3
04    ←         14    å         24    $         34    4
05    ¿         15    Ä         25    %         35    5
06    à         16    ä         26    &         36    6
07    Ø         17    Ö         27    '         37    7
08    ø         18    ö         28    (         38    8
09    ò         19    Ü         29    )         39    9
0a    ù         1a    ü         2a    *         3a    :
0b    Ň         1b    °C        2b    +         3b    ;
0c    Ç         1c    °F        2c    ,         3c    <
0d    ê         1d    ß         2d    -         3d    =
0e    É         1e    £         2e    .         3e    >
0f    é         1f    ¥         2f    /         3f    ?

40    @         50    P         60    `         70    p         
41    A         51    Q         61    a         71    q
42    B         52    R         62    b         72    r
43    C         53    S         63    c         73    s
44    D         54    T         64    d         74    t
45    E         55    U         65    e         75    u
46    F         56    V         66    f         76    v
47    G         57    W         67    g         77    w
48    H         58    X         68    h         78    x
49    I         59    Y         69    i         79    y
4a    J         5a    Z         6a    j         7a    z
4b    K         5b    [         6b    k         7b    {
4c    L         5c    \         6c    l         7c    |
4d    M         5d    ]         6d    m         7d    }
4e    N         5e    ^         6e    n         7e    ~
4f    O         5f    _         6f    o         7f    ░  

(HUI_CSET.txt)   

What are your thoughts???

masydoblig, 2019-02-20:
here is the link for the hui docs i got the character map from

masydoblig, 2019-02-20:
https://forum.cockos.com/showthread.php?t=101328

masydoblig, 2019-02-20:
https://forum.cockos.com/showthread.php?t=101328

Link for the character maps i found for HUI

masydoblig, 2019-02-20:
They are displayed in 4 characters- if there is more than 4 characters in the name it sends an f7 at the end of the message 
if there is less than 4 it adds a space as required

tttapa, 2019-02-20:
F7 indicates the end of a system exclusive message.

You would have to add a system to match the SysEx messages with the correct elements, similar to how it's done with with normal channel messages (CC, Note, ChannelPressure, etc.).

closed 2021-02-10 by tttapa