tttapa/Control-Surface #16: Transpose function

masydoblig, 2019-02-20:
Hi Pete

Does the transpose function work on rotary encoder addresses?

To change the panning from the address i set to another address 

This way i can activate/deactivate the encoder button top and adjust the lh or rh panning function in a dual pan setup?

Cheers

tttapa, 2019-02-20:
> Does the transpose function work on rotary encoder addresses?

Yes. Transpose is just a special version of Bank, so it's probably better to use Transposer for notes, and Bank for everything else.

> This way i can activate/deactivate the encoder button top and adjust the lh or rh panning function in a dual pan setup?

I don't really understand you question, but all elements have `enable` and `disable` methods.

masydoblig, 2019-02-20:
Bank should work for what I want to do
I will do some testing

Thanks again

masydoblig, 2019-02-21:
Hi Pete

i am trying to give the mackie control a try, but i cannot get the fader to work it is outputting the cc change on channel 1 address 7 but when it is mapped as a mackie control i get no fader movement in cubase 

can you check my code below to make sure it is correct with the current release, i have no idea what is going on 
i have even downloaded and installed the controlsurface library again 

#include <Encoder.h>
#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer(A0, {MIDI_CC::Channel_Volume, CHANNEL_1});
    

CCRotaryEncoder enc = {
    {2, 3},
    MCU::V_POT_1,
    2,
    4,
};

void setup() {
    // Use the Mackie Control protocol for sending relative MIDI CC messages.
    RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
    Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
    // Update the control surface
    Control_Surface.loop();
}

masydoblig, 2019-02-21:
i know it seems stupid 
i am wondering if there is a problem with the way my mackie control is working in cubase
not the controller 
i am getting a cc message correctly in midi ox 
so it should work

tttapa, 2019-02-21:
Fader movement in MCU uses Pitch Bend events, so you need a `PBPotentiometer`.

masydoblig, 2019-02-21:
That was driving me nuts
Thanks dude

masydoblig, 2019-02-23:
Hi Peter 

Just trying to get a bankable LEDs shift register working 
is this possible

Bank<2> bank(1); // Create a new bank with 1 tracks per bank

IncrementDecrementSelector<2> bankselector(bank, {8, 9}); 

ShiftRegisterOut<8> sr2(dataPin2, clockPin2, latchPin2, MSBFIRST);

Bankable::MIDINoteLED leds[] = { 
    bank,
        {{
            {sr2.pin(0), note(E, 2)}, CHANNEL_1}},
            {sr2.pin(1), {note(F, 2), CHANNEL_1}},
            {sr2.pin(2), {note(Gb, 2), CHANNEL_1}},
            {sr2.pin(3), {note(G, 2), CHANNEL_1}}, 
            {sr2.pin(4), {note(Ab, 2), CHANNEL_1}},
            {sr2.pin(5), {note(A, 2), CHANNEL_1}},
            {sr2.pin(6), {note(Bb, 2), CHANNEL_1}},
            {sr2.pin(7), {note(B, 2), CHANNEL_1}},
        }},    
};

tttapa, 2019-02-23:
Update to ee43f15 and try this:
```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

Bank<2> bank(1); // Create a new bank with 1 tracks per bank

IncrementDecrementSelector<2> bankselector(bank, {8, 9}); 

// There's no need to use different clock and data pins for each shift register,
// Either daisy chain them (i.e. use one 16-bit instance instead of two 8-bit instances)
// or share the data and clock pins, using just a different latch pin for each register.
// Daisy chaining them is preferred, see https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d82/Ex_81_8Shift-Register-LED-Effects_8ino-example.html
// It's also much more efficient and faster if you use the SPI bus for the shift 
// registers (8 MHz), and it also saves some pins.
SPIShiftRegisterOut<8> sr2(SS, MSBFIRST);

using namespace MIDI_Notes;

Bankable::MIDINoteLED<2> leds[] = { 
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(0), {note(E, 2), CHANNEL_1}},
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(1), {note(F, 2), CHANNEL_1}},
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(2), {note(Gb, 2), CHANNEL_1}},
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(3), {note(G, 2), CHANNEL_1}}, 
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(4), {note(Ab, 2), CHANNEL_1}},
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(5), {note(A, 2), CHANNEL_1}},
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(6), {note(Bb, 2), CHANNEL_1}},
    {{bank, BankType::CHANGE_CHANNEL}, sr2.pin(7), {note(B, 2), CHANNEL_1}},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

masydoblig, 2019-02-23:
Hi Peter

Tried the new version an now nothing works (tried on the Due and the Mega Boards)
tried a basic potentiometer and it doesn't even function 
when plugged into A0 it seems to short out the pins and stop communication
then when i disconnect the pot sends out a midi cc message as the voltage stabilises 

#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer(A0, {0x07, CHANNEL_1}); // working 

void setup() {
    // Initialize everything
    Control_Surface.begin();
}

void loop() {
    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
}

masydoblig, 2019-02-24:
Hi Pete

Yep it works really well 
now my fader module has a 2 bank function i can map as i please with corresponding leds 
and i have the encoder in bank 1 as "LR mono pan"  and in bank 2 it controls the "LR stereo combined panner".

I saw somewhere a bank indicator led function (to let you know which bank it is on)
but i cant seem to find it in the docs
if you could point me in the right direction that would be much appreciated 

Nice work  dude, yet again you are the man

code i have loaded at the moment below:

#include <Encoder.h>
#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

const pin_t dataPin2  = 23;  // Pin connected to DS of 74HC595 yellow wire
const pin_t clockPin2 = 25;  // Pin connected to SH_CP of 74HC595 green wire
const pin_t latchPin2 = 27;  // Pin connected to ST_CP of 74HC595 blue wire

// Create a new shift register output connected to pins 12, 13 and 10,
// shift the data out with the most significant bit first.
// There are 8 outputs in total.
SPIShiftRegisterOut<12> sr(SS, MSBFIRST);
ShiftRegisterOut<8> sr2(dataPin2, clockPin2, latchPin2, MSBFIRST);

uint8_t track = 1;
Channel channel = CHANNEL_1;
// Note: a VU meter of 12 LEDs will give the best effect, because
// in the MCU protocol, there are 12 values
// from cubase needs to be set as aftertouch on correct channel and set max value to 12 or 24
MCU::VULEDs<12> vu = { 
  {track, channel}, 
  {{
    sr.pin(0),
    sr.pin(1),
    sr.pin(2),
    sr.pin(3),
    sr.pin(4),
    sr.pin(5),
    sr.pin(6),
    sr.pin(7),
    sr.pin(8),
    sr.pin(9),
    sr.pin(10),
    sr.pin(11),
  }},
  MCU::VU::NO_DECAY, // decay time in milliseconds or MCU::VU::NO_DECAY
};

CCPotentiometer potentiometer(A0, {0x07, CHANNEL_1}); // working 

Bank<2> bank(16); // Create a new bank with 16 tracks per bank

IncrementDecrementSelector<2> bankselector(bank, {8, 9}); // does loop back to original setting from one button

Bankable::CCRotaryEncoder encoder_B = {bank, {2, 3}, 0x0A, 2, 4}; // bank 1 is normal panner, bank 2 is stereo combine panner rh. 

using namespace MIDI_Notes;
Bankable::NoteButtonMatrix<4, 4> keypad = {
    bank,
    {22, 24, 26, 28},                         
    {30, 32, 34, 36}, // Column pins (input-pullup)
    // All notes
    {{
        {          
            note(C, -1), // MUTE BUTTON
            note(Db, -1), // SOLO BUTTON
            note(D, -1), //
            note(Eb, -1),
        },
        {   
            note(E, -1), // READ BUTTON
            note(F, -1), // WRITE BUTTON
            note(Gb, -1),
            note(G, -1),
        },
        {    
            note(Ab, -1), // MONITOR BUTTON
            note(A, -1), // RECORD BUTTON
            note(Bb, -1),
            note(B, -1),
        },
        {   
            note(C, 0), //INSERT BYPASS
            note(Db, 0), //SENDS BYPASS
            note(D, 0),
            note(D, 0),
        },

    }},
};

Bankable::MIDINoteLED<2> leds[] = { 
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(0), {note(E, 2), CHANNEL_1}}, // RECORD LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(1), {note(F, 2), CHANNEL_1}}, // WRITE LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(2), {note(Gb, 2), CHANNEL_1}}, // SOLO LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(3), {note(G, 2), CHANNEL_1}}, // MUTE LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(4), {note(Ab, 2), CHANNEL_1}}, // READ LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(5), {note(A, 2), CHANNEL_1}}, // MONITOR LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(6), {note(Bb, 2), CHANNEL_1}}, // INSERT BYPASS LED
    {{bank, BankType::CHANGE_ADDRESS}, sr2.pin(7), {note(B, 2), CHANNEL_1}}, // SEND BYPASS LED
};

void setup() {
    // Initialize everything
    Control_Surface.begin();
}

void loop() {
    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
}

tttapa, 2019-02-24:
> I saw somewhere a bank indicator led function (to let you know which bank it is on)
but i cant seem to find it in the docs
if you could point me in the right direction that would be much appreciated

You can use [`IncrementSelectorLEDs`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/dc/df8/classIncrementSelectorLEDs.html). (There's `IncrementDecrementSelectorLEDs` as well, but it's not really useful if you only have 2 banks.)

masydoblig, 2019-02-24:
Ah that was it cheers mate I will add it to the code
It will show me which fader modules are in the stereo combined panner mode
Thanks again you are a champion

cferrarini, 2020-09-08:
I noticed that the transposer example, lets only the user choose the +or- semitones limit , but the sketch coud also have the "steps" variable, in it, in the case he wants to transpose a whole octave at each +or- transposer button press.
In my case, my controller has only 8 note buttons, so I changed the steps to 8 in the library, so I can cover the whole keyboard. Supposing the user had a 1 ocatve keyboard, the step need to be 12,
SKETCH -> Transposer<-6, +6> transposer;
Library _> Transposer(int8_t step = 8)

closed 2021-02-10 by tttapa