tttapa/Control-Surface #17: Possibility of communication between modules

masydoblig, 2019-02-24:
Hi Pete

Is there a possibility of getting the Arduino boards to communicate with each other so that each module is on the buss with one master bus passing the Midi information to/from the PC.
Either over I2C or similar using the spare TX RX ports

If the message is a channel 2 message it only gets recognized by the #2 module

The modules can be programmed exactly the same except for the midi channel number 
All of the slaves would only need to be powered by a 9v supply.and connected on the bus.

i.e. 

Fader module 1 : (mega 1 sends and receives midi data to/from Master)
CCPotentiometer potentiometer(A0, {MIDI_CC::Channel_Volume, CHANNEL_1});

Fader Module 2 :  (mega 2 sends and receives midi data to/from Master)
CCPotentiometer potentiometer(A0, {MIDI_CC::Channel_Volume, CHANNEL_2});

Master Module: (Mega 16) Passes all information to PC
CCPotentiometer potentiometer(A0, {MIDI_CC::Channel_Volume, CHANNEL_16});

I am about to start the design of the EQ Module and Master Modules
All will have their own controller(Mega) and custom pcb 
[image]

tttapa, 2019-02-24:
I've thought about this, but I haven't implemented it yet (and I can't test it at the moment).

I²C seems an easy solution:
- Master receives MIDI over USB, parses it itself, and sends it over I²C to all slaves
- Slaves receive MIDI over I²C and parse it.
- Master sends MIDI over USB
- Slaves store outgoing MIDI in the I²C buffer
- Master requests outgoing MIDI from slaves and sends it over USB

There are a couple of caveats, however:
- I don't know if the standard Wire library supports broadcast
- Slaves can only send complete MIDI messages when the master requests it, if the slave is in the process of writing a long message into the buffer, and the I²C interrupt fires, it may send an incomplete MIDI message, then after that, MIDI messages from other slaves are sent over the same MIDI USB interface, causing malformed MIDI packets, or just mixing them up.  
The solution seems to be to make all slave I²C buffer writes atomic, by disabling interrupts when writing to the buffer.
- How to prevent SysEx messages from being interrupted by data from other slaves? I don't know if I can make this atomic.
- Is the 32-byte Wire buffer large enough?
- If the master uses MIDI over USB, the I²C data to and from the slaves must be USB packets. If the master uses Serial MIDI, the I²C data must be Serial MIDI as well. 
If the protocols don't match, MIDI USB Cable Number data will get lost, and resources are wasted converting from one protocol to another.

masydoblig, 2019-02-24:
Could the sysex be disabled from being sent on the I2C bus and only recognized by the master? If required.

As you have said before the mega uses TX0 and RX0 for the communication so
So could the  midi transfer bus be on TX2 RX2?
To avoid the interrupt issues? And would that then mean it could be in a midi serial protocol.

I'm sure this is a shit load of work to get it working but it would make your system absolutely powerful as hell and completely modular there could be 1 to 32 modules (16 faders 16 eq) on a single usb as there is dfinately enough midi assignments avaliable

.

tttapa, 2019-02-24:
> So could the midi transfer bus be on TX2 RX2? To avoid the interrupt issues? And would that then mean it could be in a midi serial protocol.

That's not possible, because only one slave can send data on the UART, you would need some kind of external bus access control, which is even more complicated, and should also use interrupts for reliability.  
I²C is better suited because it handles the bus management for you.  
You can perfectly send serial MIDI over I²C, that's not the issue here.

> I'm sure this is a shit load of work to get it working but it would make your system absolutely powerful as hell and completely modular there could be 1 to 32 modules (16 faders 16 eq) on a single usb as there is dfinately enough midi assignments avaliable

I'm afraid I don't have the time right now. I have to focus on my studies, and adding this feature would take up too much time. It's not just writing code, you need architectural design, hardware testing, unit tests, refactoring, documentation, examples, etc.  
It's too much work for the limited time I have available.

I'm thinking it might be easier to just use the ExtIO functions with "dumb" IO expanders instead of extra Arduinos.

masydoblig, 2019-02-24:
I have been meaning to ask what are you studying?

tttapa, 2019-02-25:
I'm in my third year of Electrical Engineering and Computer Science.

masydoblig, 2019-02-25:
That explains the awesomeness at control surface programming.

masydoblig, 2019-02-28:
Hi Pete

when you get time would you be able to help me with some code using the I2c 
i have no idea how to do it (i am going to have a look on the weekend)

i would like to have 2 buttons on the master module that when pressed send out a message to the slaves and the slaves will send a note on off to the pc (to change the vst track up or down in cubase as i can program this in the daw) 
if this is possible 

that way i can set up the fader modules i have with their own usb port and when they get the signal from the master they can all shift to their nex designated track at the same time in cubase
ie fader 1 would go to fader 17 (if i had 16 fader modules built)

Cheers Sam

tttapa, 2019-03-01:
Is there a specific reason you're not using the ExtIO classes for buttons and LEDs? I think that would be much easier.

I²C is pretty straightforward, the master sends a message to a slave on a specific address, and the slave can reply to that message. The Arduino site has some working examples.  
The next problem is coming up with some kind of protocol to tell the slave what to do. Then you'll also need a parser for that protocol.  
My repository with motor fader code I linked to earlier contains an example of this (to set the parameters of the fader controller).

masydoblig, 2019-03-01:
I am using the ext io functions for the push buttons and the LEDs on each module.

but with 16 faders each having 10 buttons (cubase specific) and a rotary encoder for each fader i would run out of pins quite quickly even with the ext io. 
not to mention the eq module would have 16 rotary encoders and buttons
or the implementation of the motor faders .

Within cubase if i plug in one usb controller i can assign it to a generic remote
[image]
I currently have 2 fader modules plugged in on 2 generic controllers 

Then in the editor for each generic controller i can create and delete custom vst banks which i can make anything i want it to be 
i.e in a 16 fader module set up fader 1 module has 5 vst banks
VST 1 Channel 1 
VST 2 Channel 17
VST 3 Channel 33
VST 4 Channel 49
VST 5 Input Channel 1
[image]

But to get all of the faders to change vst banks simultaneously i need to get them all to send out a midi message to the daw (button press from master module over the i2c) or connect all 16 pins assigned to a note button on each controller to a single switch and ground them at the same time (could cause issues).

This is why i was asking about the I2C comunication of midi data between modules so i can program cubase as one controller sending messages on different channels which makes the shifting of vst's easier

What are your thoughts?
also i understand the stress you would be under from university and that this is a side issue. so please dont feel obligated.

Thanks In advance Sam
My face time is mas.ydoblig@hotmail.com if you would like to see my setup and discuss the options in more depth.

masydoblig, 2020-10-16:
Hi Peter 

It looks like you have implemented the communication between master and slave is this correct?

Does this mean i can have the master connected to the DAW and send midi data to the slaves like my examples above?
if so how do i do it? @tttapa 

Cheers Sam

tttapa, 2020-10-17:
Not directly, but you could use the MIDI Routing module to achieve something like this:

- https://tttapa.github.io/Control-Surface-doc/Doxygen/df/ddc/group__MIDI__Routing.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/df/d72/classMIDI__Pipe.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da5/Dual-MIDI-Interface_8ino-example.html

You'll have to implement the `TrueMIDI_Sink` interface to send MIDI to your slave, and `TrueMIDI_Source` to receive MIDI from your slave. How these implementations work is up to you, you can use pretty much any protocol you want.

closed 2021-02-10 by tttapa