tttapa/Control-Surface #20: MMC

eurodvd, 2019-05-12:
Hi,
I am trying to make a simple remote control with MIDI, just play, stop, rec button with feedback from the controlled machine, lighting led for each function. I have tried your library but no way. I am trying to control protools with hui protocol. It seems that MCU does not match HUI. 
Do you have a clue?
Thanks?
Pierre

eurodvd, 2019-05-12:
Hi Piete,
Ok I have just tried with reaper, and everithing works perfectly.
Though my problem stay as I need to make it working for Prtools with HUI protocol. Do you have an idea, do you think it is possible to just change the commands value?
Thank you.
Pierre

tttapa, 2019-07-25:
My apologies for the late reply.

I think the command values are just different, but there's not that much information available about the protocol. I think I have some files on an old HDD that describe part of the protocol.

I'll post it here if I can find them.

tttapa, 2019-07-25:
Here are some links to the files I was talking about: <https://forum.cockos.com/showthread.php?s=52a01e090fb5540370effda832142f09&t=101328>

It's not at all easy to implement, because it uses multiple events to select a zone and then send the actual switch/LED.

closed 2019-10-09 by tttapa