tttapa/Control-Surface #21: Potentiometer Calibrated? Need Help

lucky18perdana, 2019-06-16:
There is a example for Potentiometer map Calibrated like the old version? Thank you

tttapa, 2019-06-16:
I added an example here: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html>

LuckyMann, 2019-06-20:
Thank You, I'll try it.

LuckyMann, 2019-06-20:
Its working. but unfortunately I didnt get smooth result like the old Libraries. So Im Back to the old one. Thanks anyway.

tttapa, 2019-06-20:
Could you post your mapping function?

The difference with the old one is that the new one reduces the bit depth first, before applying the mapping function. This reduces the amount of noise, but comes at a cost of losing bit depth for the mapping.  
Do you get a lot of noise with the old code, or is it not an issue? If not, I might change it back in the new library.

LuckyMann, 2019-06-23:
I do exactly as the example method. And the “get raw value” doesnt work too. Or maybe just me. 

I dont know about the others, but I feel getting best smooth callibration result when using the old library.

commit 31edaa6b (2019-08-09): Fix #21: apply map function before hysteresis
closed 2019-08-09 by tttapa