tttapa/Control-Surface #22: Espressif chips

chaffinched, 2019-06-26:
I was looking for a way to revive old controllers and this looks ideal. Are there any plans to get this working for ESP8266 or the bigger ESP32? They have wireless and BLE onboard along with loads of GPIO and may be cheaper than their Arduino equipvalents.

tttapa, 2019-06-26:
There is some MIDI BLE code for ESP32. I think it works, but it's quite experimental, and I haven't had time yet to clean it up, wrire documentation, or add examples.

<https://tttapa.github.io/Control-Surface/Doc/Doxygen/d0/d89/classBluetoothMIDI__Interface.html>

chaffinched, 2019-06-26:
If you think the rest of the code will work on an ESP32 I'll try it out and let you know how I get on. I really need to stop buying faulty midi keyboards on ebay, but if I can give them a new life it will be worth it!

chaffinched, 2019-06-26:
Getting-started.ino compiles for me for an Arduino Uno, but for any ESP32 boards I get this:

```c++
WARNING: library Control-Surface claims to run on (avr, sam, samd, teensy, esp) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
In file included from C:\Users\username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/esp32-hal.h:53:0,

               from C:\Users\username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:35,

               from sketch\Getting-Started.ino.cpp:1:

C:\Users\username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/esp32-hal-gpio.h:51:19: error: expected unqualified-id before numeric constant

#define DISABLED  0x00

                 ^

C:\Users\username\Documents\git\Arduino\libraries\Control-Surface\src/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp:132:30: note: in expansion of macro 'DISABLED'

   constexpr static uint8_t DISABLED = HIGH;

                            ^

exit status 1
Error compiling for board ESP32 Dev Module.
```

I tried a couple of the other examples and got variations of that error. Is it just me?

tttapa, 2019-06-26:
Apparently, the new version of the ESP32 defines a public macro that conflicts with some internal variables of the Control Surface library. I already fixed it in 557097532f3e96071e0c6945f8301324605b4636, so you can just pull the latest version from GitHub. 

I also fixed some other BLE things, and I just tested it on Ubuntu 16.04 (Bluez 5.50) and Android 9.0.5.

chaffinched, 2019-06-26:
Wow, that's fast work! I can compile the examples that I tried, thanks!

tttapa, 2019-06-26:
There are now three basic BLE MIDI examples. Here's the one that demonstrates both input and output:
<https://tttapa.github.io/Control-Surface/Doc/Doxygen/d4/d4d/Bluetooth-BLE-MIDI-NoteButton_Input_8ino-example.html>

chaffinched, 2019-07-04:
Sadly, I haven't been able to get my Windows 10 laptop to recognise the BLE connection as a MIDI device. I'll close this issue as it seems to be a problem my end rather than with your code, as I've had the same problem with other people's code. USB will have to do for me! Thanks.

closed 2019-07-04 by chaffinched