tttapa/Control-Surface #26: The lib is not compiled

GooseDB, 2019-07-24:
src/MIDI_Inputs/MCU/VU.hpp:14:31: error: macro "max" requires 2 arguments, but only 1 given
     IVU(uint8_t max) : max(max) {}

https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Inputs/MCU/VU.hpp#L14

I can't get what this code does, and can't fix it myself. Can you fix?

tttapa, 2019-07-24:
What board are you using?

The reason is that the Arduino core for your particular board defines a global max macro, which is bad practice because it gives rise to name collisions.

tttapa, 2019-07-25:
I created a new branch with a temporary fix. Does it compile now, or are there other errors?

GooseDB, 2019-07-25:
Yes. You are right. Thank you.

GooseDB, 2019-07-25:
Yes. It works. Thank you.

closed 2019-07-25 by GooseDB