tttapa/Control-Surface #29: Getting MIDI Clock from DAW

ransalman, 2019-08-02:
Hi
First of all thank you for this great library.
I have a sketch that works fine with getting the realtime messages via midi.read() / serial.read() and sync is working fine but once I call Control_Surface.begin() the midi clock timing is messed up.
I saw that there is a "onRealtimeMessage" callback in the documentation but I can't manage to get it to work, can you please share an example on how to define callback for real time midi messages in a sketch?
Thank you.

tttapa, 2019-08-02:
The callback is in the Control_Surface_ class. There's no user-accessable callback at thr moment, but I'll add one tonight.

ransalman, 2019-08-02:
amazing, thank you

tttapa, 2019-08-02:
Added it in e786d11ba5c2be0be7c7da30a275ffad4120b886

You can use it like this:
```cpp
#include <Control_Surface.h> // Include the Control Surface library

USBDebugMIDI_Interface midi;

bool realTimeMessageCallback(uint8_t message) {
    Serial << "Real-Time message: " << hex << message << dec << endl;
    return true;
}

void setup() {
    Serial.begin(115200);
    Control_Surface.begin();
    Control_Surface.setMIDIInputCallbacks(nullptr, 
                                          nullptr, 
                                          realTimeMessageCallback);
}

void loop() {
    Control_Surface.loop();
}
```

Documentation here: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/dc/d45/classControl__Surface__.html#ac9f7e995df0b998287da442587f73e1b>

ransalman, 2019-08-04:
Works amazing, thank you very much

tttapa, 2019-08-04:
I'm going to reopen it for now, because I'll have to add the MIDI Cable Number as a parameter.

tttapa, 2019-08-06:
I added the Cable Number in f9455f87a29d26e28f9c028514e5854930849795.

https://github.com/tttapa/Control-Surface/blob/f9455f87a29d26e28f9c028514e5854930849795/examples/Other/MIDI_Callback/MIDI_Callback.ino#L1-L24

closed 2019-08-06 by tttapa