tttapa/Control-Surface #31: Invert Potentiometer

W-Floyd, 2019-08-12:
I looked, but I don't see a way to invert a potentiometer signal. Is this possible to do? I would rather not need to reverse polarity in hardware. Specifically, I have an array of potentiometers.

tttapa, 2019-08-12:
```cpp
CCPotentiometer pots[] = {
  {A0, MIDI_CC::General_Purpose_Controller_1},
  {A1, MIDI_CC::General_Purpose_Controller_2},
  {A2, MIDI_CC::General_Purpose_Controller_3},
  {A3, MIDI_CC::General_Purpose_Controller_4},
};

analog_t invertPot(analog_t val) {
  return 1023 - val;
}

void setup () {
  for (auto &pot : pots)
    pot.map(invertPot);
    Control_Surface.begin();
}
```

I can't test it right now, and I've been changing some things about the mapping function lately, so please let me know if it works.

W-Floyd, 2019-08-12:
That works perfectly! Thanks so much.

closed 2019-08-12 by W-Floyd

tttapa, 2019-08-12:
Glad to hear!

benwadub, 2020-06-28:
hi, I got the same problem, I made my pub and inverted the way they should work, and on top and vcc on bottom :-(
could you tell me how I could invert it 
*cpp
Bankable::CCPotentiometer fader [] = { //ici on déclare les faders avec comme vu plus haut le numéro de CC et le canal
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(8), {7, CHANNEL_1}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(9), {7, CHANNEL_2}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(10), {7, CHANNEL_3}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(11), {7, CHANNEL_4}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(12), {7, CHANNEL_5}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(13), {7, CHANNEL_6}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(14), {7, CHANNEL_7}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(15), {7, CHANNEL_8}}, 
};

W-Floyd, 2020-06-28:
Though I no longer needed the inverted function due to corrected wiring, have a look at this

https://github.com/W-Floyd/midiMixer/blob/master/v1/code/src/main.cpp

You can replace the invert function with the one given by https://github.com/tttapa/Control-Surface/issues/31#issuecomment-520625143

tttapa, 2020-06-28:
The changes to the mapping functions are mentioned in the readme:

>  - 31edaa6b76477fdf152c19fd34f7e4e8506561e6  
   The **mapping function** is now applied before applying hysteresis.  
   This means that the input and output values of the function should be 
   16 - `ANALOG_FILTER_SHIFT_FACTOR` bits wide instead of 7. By default this is
   **14 bits**.  
   The signature of the mapping function is now `analog_t f(analog_t raw)`, 
   where the return value and raw are both numbers in [0, 16383].

For general mapping, see this example: [**CCPotentiometer-Map.ino**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html)  
Inverting the potentiometer is supported directly using the [**`CCPotentiometer::invert`**](https://tttapa.github.io/Control-Surface-doc/Doxygen/db/d32/classCCPotentiometer.html#a7fa1616cc61c19a5efcc863c950f7f30) function:

```cpp
CCPotentiometer potentiometer = {
    A0, {MIDI_CC::Channel_Volume, CHANNEL_1}
};
 
void setup() {
    potentiometer.invert();
    Control_Surface.begin();
}
```
Of if you have an array of potentiometers, use a for loop:
```cpp
Bankable::CCPotentiometer faders[] = {
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(8), {7, CHANNEL_1}},
  {{bank3, BankType::CHANGE_CHANNEL}, mux3.pin(9), {7, CHANNEL_2}},
  // ...
};

void setup() {
    for (auto &fader : faders)
        fader.invert();
    Control_Surface.begin();
}
```

benwadub, 2020-06-29:
Thanks a lot! With the for loop will it only invert my faders and let the other pots in the normal functionement?