tttapa/Control-Surface #32: Shift Registers with leds

kentforth, 2019-08-22:
Hi Pieter!
I want to use shift register 74HC595 to connect 8 leds to Arduino Leonardo. I saw your example TOGGLE LEDs, but I didn't find line with code where you define Arduino pins for connecting shift registers. Shift register require 3 pins for communication with Arduino(latch pin, data pin and clock pin) I just can't find where you define it.

May be there is some file in your library where you did it?

Can you explain this to me?

kentforth, 2019-08-22:
The problem is I don't understand what pins on Arduino are MOSI, CLK and SS?  if I replace SS by any digital arduino pins like that:
SPIShiftRegisterOut<mux.length()> sreg = {10, MSBFIRST};

What about other pins MOSI and CLK? Where should I connect that pins on Leonardo?

tttapa, 2019-08-22:
The SPI pins are on the ICSP header.

[image]

kentforth, 2019-08-23:
I found only MOSI pin here. Can you highlight other 2 pins?
Can SS pin be any digital pin on arduino?

tttapa, 2019-08-23:
CLK (clock) and SCK (serial clock) are the same. In master mode (which is what you use for shift registers), SS can be any digital pin.

kentforth, 2019-08-23:
Am I right?

[image]

tttapa, 2019-08-23:
I think CLK is the top one in the middle, not the right one.

And SS can be any pin, as long as you change it in the code. Pin 10 is a popular choice, because it's the SS pin of the Arduino UNO.

kentforth, 2019-08-24:
Thanx for your answer.
I appreciate your work! I will close issue

closed 2019-08-24 by kentforth

tttapa, 2019-08-24:
Glad to hear :)