tttapa/Control-Surface #33: Teensy 3.6 Problem with Mux (CD74HC4067) and Reading Buttons

ransalman, 2019-08-22:
I have a problem that drives me crazy, same wiring and code working 100% on Arduino Mega but on Teensy 3.6 one button affects the other once.

I tried reducing the Teensy CPU Speed to 24MHZ and it looks like it is working correctly,
I searched in the documentation and didn't see a special settings to define once using Teensy...
Am I missing something?

another less urgent question - how can I change the midi device name ? tried renaming the string in Teensy-USB-MIDI-name.c and it didn't work...

Thanks

tttapa, 2019-08-22:
The problem is that the Teensy is too fast for the multiplexers, resulting in cross talk. 

You can add a delay of 5 µs to the code that drives the address lines.
https://github.com/tttapa/MIDI_controller/issues/87

I'll try to add it on a separate branch so you can test if that solves your problem.

ransalman, 2019-08-22:
Thanks for the fast reply
Problem is I'm using it as a step sequencer (midi clock slave)
I'm afraid the delay will break the midi sync, or it won't?

tttapa, 2019-08-22:
5 microseconds are not noticeable, and it's negligible compared to the other code that's running.  
As a comparison, sending a single MIDI Note event on a serial DIN MIDI interface takes 1000 µs, for example.

tttapa, 2019-08-22:
I added a new branch with the fix. Could you try it? Then I'll merge it into master.
https://github.com/tttapa/Control-Surface/tree/mux-delay-teensy

Do you know how to use git to checkout this branch? Did you install the library as a ZIP file, or did you use git to clone it?

tttapa, 2019-08-22:
> another less urgent question - how can I change the midi device name ? tried renaming the string in Teensy-USB-MIDI-name.c and it didn't work...

Not sure, maybe it takes some time (or a reboot) for your computer to recognize the new name? It probably caches it, since it's unusual to change USB device names while it's in use.  
I haven't tried it with a Teensy 3.6 yet, but it seems to update immediately with my Teensy 3.2 on Ubuntu 16.04.

ransalman, 2019-08-22:
> Control Surface

> By default, does your device show up as "Control Surface", or something else?

Yes,need to restart and check

ransalman, 2019-08-22:
> I added a new branch with the fix. Could you try it? Then I'll merge it into master.
> https://github.com/tttapa/Control-Surface/tree/mux-delay-teensy
> 
> Do you know how to use git to checkout this branch? Did you install the library as a ZIP file, or did you use git to clone it?

yes, I checked it out.. there is an improvement of buttons reading but there are still issues. I think the delay should be a bit more.. I can play with the delayMicroseconds value you put there and report the results here.
Thanks man

ransalman, 2019-08-26:
I was wrong, I had another problem and the 5 µs indeed helped !
Thanks

BTW - Changing the device name in Teensy-USB-MIDI-name.c is only updated on another Mac...  Guess it's something to do with my Mac caching the device name.. not urgent at all

tttapa, 2019-08-27:
Glad to hear you got it to work!

closed 2019-08-27 by tttapa

benwadub, 2021-11-06:
Hi, I have a similar problem on 2 mux, my big project using 10 cd4067 as this problem on my mux only on push button latched, no problem on the 124 potentiometer do you know what can cause this problem only on push button but not on pots?