tttapa/Control-Surface #34: How to use Midi CC instead of Note with button?

kentforth, 2019-08-27:
Hi Pieter!
I want to use button and send midi cc, but in your example : Midi Output/Buttons and Switched/Momentary push buttons/NoteButton you made sending only NoteNumber

**How can I change the code to send Midi CC to my DAW instead of NoteNumber with a button?**

**Can you give me a nexample of using a button that sends midi cc?**

tttapa, 2019-08-27:
You can find an overview of all MIDI elements in the documentation:  
https://tttapa.github.io/Control-Surface/Doc/Doxygen/d7/dcd/group__MIDIOutputElements.html

You can use `CCButton` instead of `NoteButton`.

kentforth, 2019-09-29:
I didn't find an example for button that sends MIDI CC instead of Midi Note, can you show a simple sketch where pressing a button will send midi cc?

tttapa, 2019-09-29:
I added an example: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d9/d2e/CCButton_8ino-example.html

commit 33134ed6 (2019-09-29): Add CCButton example
close #34
closed 2019-09-29 by tttapa