tttapa/Control-Surface #35: Dual midi interface

pandeagus, 2019-09-03:
hello Pieter
i got error using dual midi interface... im using lilypad atmega32u4 
the problem is when i upload dual midi interface always got error Couldn't find a Board on the selected port. but if i just using USBMIDI_Interface midi always succesfull.

and this my code.. 

#include <Control_Surface.h>

// Instantiate a MIDI over USB interface.
//USBMIDI_Interface midi;

USBMIDI_Interface usbmidi;
HardwareSerialMIDI_Interface serialmidi = {Serial1, MIDI_BAUD};
MultiMIDI_Interface<2> midi = {{&usbmidi, &serialmidi}};

// Instantiate the momentary pbuush buttons that send out MIDI Note events.
CCButtonLatched  buttons[] = {
    {21, MIDI_CC::Effects_1}, // 
    {19, MIDI_CC::Effects_2},
};

constexpr size_t numberOfButtons = sizeof(buttons) / sizeof(*buttons);

// The pins of the LEDs that will display the state of the buttons
pin_t ledPins[] = {
    18,
    15,
};

constexpr size_t numberOfLEDs = sizeof(ledPins) / sizeof(*ledPins);
static_assert(numberOfButtons == numberOfLEDs,
              "Error: number of buttons is not "
              "the same as the number of LEDs!");
void setup() {
    // Initialize everything
    Control_Surface.begin();

    for (pin_t &pin : ledPins)
        pinMode(pin, OUTPUT);
}

void loop() {
    // Update the control surface: read new MIDI events, and read all inputs
    Control_Surface.loop();

    for (uint8_t i = 0; i < numberOfLEDs; ++i)
        digitalWrite(ledPins[i], buttons[i].getState());
}

tttapa, 2019-09-03:
That's strange. I'll try to test it tonight, I don't have any ATmega32U4s at hand right now.

pandeagus, 2019-09-03:
okay i will be waiting..  i just want use midi over usb and 5-pin DIN MIDI.
thanks for respones

tttapa, 2019-09-03:
I haven't been able to reproduce the "Couldn't find a Board ..." error, maybe because I'm on Linux, but I have fixed the bug where MultiMIDI_Interface doesn't initialize its interfaces.

391fa167a854d7db03d9ce587d8ad9a6a01d11db

pandeagus, 2019-09-04:
[image].

still error  i dont know why.
i  want ask how do i use the 5 din pin midi...
what interface code should i use?
HardwareSerialMIDI_Interface midi;
 or 
StreamMIDI_Interface midi;
or
SerialMIDI_Interface midi;

tttapa, 2019-09-04:
That's a different error. Please post the exact code and error as text, not as an image.

You need `HardwareSerialMIDI_Interface`.

closed 2019-10-04 by tttapa