tttapa/Control-Surface #37: button matrix.ino

brorgustav, 2019-09-07:
With the new "fix" it seems to compile the basic examples but the button-matrix sketch is still giving me error in compile: 

Ex.11.Button-Matrix:34:81:` error: no matching function for call to 'ButtonMatrix<4, 3>::ButtonMatrix(<brace-enclosed initializer list>, <brace-enclosed initializer list>, const uint8_t [4][3], int, const uint8_t&)'
 ButtonMatrix<4, 3> buttonmatrix( {2, 3, 4, 5}, {6, 7, 8}, addresses, 1, velocity);
                                                                                 ^
In file included from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.hpp:78:0,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/MIDI_Outputs/Abstract/MIDIButtonMatrix.hpp:4,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/MIDI_Outputs/CCButtonMatrix.hpp:3,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Control_Surface.h:31,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/examples.old/very-old/Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:22:
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.ipp:8:1: note: candidate: ButtonMatrix<nb_rows, nb_cols>::ButtonMatrix(PinList<nb_rows>&, PinList<nb_cols>&) [with unsigned char nb_rows = 4; unsigned char nb_cols = 3; PinList<nb_rows> = Array<short unsigned int, 4>; PinList<nb_cols> = Array<short unsigned int, 3>]
 ButtonMatrix<nb_rows, nb_cols>::ButtonMatrix(const PinList<nb_rows> &rowPins,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.ipp:8:1: note:   candidate expects 2 arguments, 5 provided
In file included from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/MIDI_Outputs/Abstract/MIDIButtonMatrix.hpp:4:0,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/MIDI_Outputs/CCButtonMatrix.hpp:3,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Control_Surface.h:31,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/examples.old/very-old/Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:22:
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.hpp:18:7: note: candidate: constexpr ButtonMatrix<4, 3>::ButtonMatrix(const ButtonMatrix<4, 3>&)
 class ButtonMatrix {
       ^~~~~~~~~~~~
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.hpp:18:7: note:   candidate expects 1 argument, 5 provided
Ex.11.Button-Matrix:34:20: error: cannot declare variable 'buttonmatrix' to be of abstract type 'ButtonMatrix<4, 3>'
 ButtonMatrix<4, 3> buttonmatrix( {2, 3, 4, 5}, {6, 7, 8}, addresses, 1, velocity);
                    ^~~~~~~~~~~~
In file included from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/MIDI_Outputs/Abstract/MIDIButtonMatrix.hpp:4:0,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/MIDI_Outputs/CCButtonMatrix.hpp:3,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Control_Surface.h:31,
                 from /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/examples.old/very-old/Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:22:
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.hpp:18:7: note:   because the following virtual functions are pure within 'ButtonMatrix<4, 3>':
 class ButtonMatrix {
       ^~~~~~~~~~~~
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/src/Hardware/ButtonMatrix.hpp:63:18: note: 	void ButtonMatrix<nb_rows, nb_cols>::onButtonChanged(uint8_t, uint8_t, bool) [with unsigned char nb_rows = 4; unsigned char nb_cols = 3; uint8_t = unsigned char]
     virtual void onButtonChanged(uint8_t row, uint8_t col, bool state) = 0;
                  ^~~~~~~~~~~~~~~
/Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface/examples.old/very-old/Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino: In function 'void loop()':
Ex.11.Button-Matrix:40:19: error: 'class Control_Surface_' has no member named 'refresh'
   Control_Surface.refresh();
                   ^~~~~~~
Multiple libraries were found for "SPI.h"
 Used: /Users/brorgustav/Documents/Arduino/Sketches/libraries/SPI
 Not used: /Users/brorgustav/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/libraries/SPI
Using library Control-Surface at version 1.0.0 in folder: /Users/brorgustav/Documents/Arduino/Sketches/libraries/Control-Surface 
Using library SPI at version 1.0 in folder: /Users/brorgustav/Documents/Arduino/Sketches/libraries/SPI 
Using library MIDIUSB at version 1.0.4 in folder: /Users/brorgustav/Documents/Arduino/Sketches/libraries/MIDIUSB 
exit status 1
no matching function for call to 'ButtonMatrix<4, 3>::ButtonMatrix(<brace-enclosed initializer list>, <brace-enclosed initializer list>, const uint8_t [4][3], int, const uint8_t&)'

tttapa, 2019-09-07:
You seem to be compiling an old example that's not compatible with the new version of the library.

See the documentation of NoteButtonMatrix here: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d5/df9/group__MIDIOutputElementConstructors.html#ga23e45eeee7965faa832397bc1a842446

For example:

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

// The note numbers corresponding to the buttons in the matrix
const AddressMatrix<4, 3> addresses = {{
  {  1,  2,  3 },
  {  4,  5,  6 },
  {  7,  8,  9 },
  { 10, 11, 12 }
}};

NoteButtonMatrix<4, 3> buttonmatrix = {
  {2, 3, 4, 5}, 
  {6, 7, 8}, 
  addresses, 
  CHANNEL_1,
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

brorgustav, 2019-09-07:
Thank you for support, I got it working! Your code runs better than my matrix scan code, mine has bugs. I will use this for my midi keyboard ty! :)

brorgustav, 2019-09-07:
Is there a max polyphony/simultaneous presses?

brorgustav, 2019-09-07:
Now the question is how I would go about to make a velocity interface matrix for the velocity buttons on the PCB. Is there anyway to implement another matrix that doesn’t execute anything like USBMIDI?

tttapa, 2019-09-07:
What do you mean by "velocity buttons"?

You don't have to use the NoteButtonMatrix class. You can use the ButtonMatrix class that just reads the buttons, without any MIDI stuff (it is the base class that is used by the MIDI Button Matrix classes).

https://tttapa.github.io/Control-Surface/Doc/Doxygen/d8/d6c/classButtonMatrix.html

If that doesn't fit your needs, you can just write your own classes, you can easily add MIDI Elements to the Control Surface library, and you can use the utilities it provides, like the MIDI functions and the hardware elements.

tttapa, 2019-09-07:
> Is there a max polyphony/simultaneous presses?

No, there's no limit. (Polyphony limits are usually because of the sampler that turns MIDI into sound, it can only play and mix a limited number of samples at the same time.)

However, as with all scanning button matrices, you can only press multiple keys at once if every key has a diode installed.

closed 2019-10-04 by tttapa