tttapa/Control-Surface #38: botton bank

zley1, 2019-09-30:
hi, i am having trouble on a code for a 6 banks of buttonmatrix with 6 leds and two butons for changing banks
buttom matrix is 6 buttons (two rows 3 colums)
i want to be able to choose the note for each button in every bank. i try some of the examples but i get errors with them.
thanks in advance

tttapa, 2019-09-30:
I added an example, but I haven't tested it yet.

zley1, 2019-09-30:
thnx!! but how do i choose the notes for each bank? i want to have different scales for playing melodies.

tttapa, 2019-10-01:
That's currently not supported. But I'm in the process of adding the code for it.

zley1, 2019-10-02:
ok thnx!! let me know if u make it.

tttapa, 2019-10-25:
I added [NoteButtonMatrix](https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d39/classBankable_1_1ManyAddresses_1_1NoteButtonMatrix.html), but I don't have time at the moment to add the LEDs.

panakronic, 2019-12-16:
Hi 

I have 4 banks with two butons for changing banks. Please see example MIDI_controller-97.

My question: Is is possible to use RGB LED strip with 4 pre-defined colors instead of 4 seperate LED of different color? I am currently using a 5mm 4-pin RGB LED bulb and it allows me only 3 color changes - Red, Green and Blue. How can I add one or more colors?

tttapa, 2019-12-16:
The LED outputs for the Selectors are defined herehttps://github.com/tttapa/Control-Surface/blob/aa4086fe1c93ba66bf510fef3485a82a0e5202ef/src/Selectors/LEDs/SelectorLEDs.hpp#L9-L33

You can easily write your own callback to drive the LEDs the way you want. Then you just pass your callback class to the Generic***Selector template classes, as shown in further down in the file mentioned earlier.

closed 2021-02-10 by tttapa