tttapa/Control-Surface #39: Pro Micro Supported ?

orzdk, 2019-09-30:
Hi Pieter
Thank you for your work on this great library ! 

I need to ask - Is Pro Micro ATMega32U4 currently supported? 

I tried the Potentiometer example and code compiles and uploads fine, but USB device is not recognized in Windows.  This is the same for Control-Surface as well as MIDIController. The unit should be ok, USBMidiKlik seems to be recognized. 

This could very well be problems with my toolchain, I just wanted to check if it's supposed to work : )

Thank you a million in advance for clarifying this, and have a nice day.
BR Jesper

tttapa, 2019-09-30:
I have never tested the Pro Micro. The official Arduino Leonardo with the same microcontroller is supported, but I assume that the core USB firmware of the Pro Micro is different.

USB MIDI support on the ATMega32U4 is handled by the MIDIUSB library, so I suspect that the problem lies there. Could you try some of the examples that come with the MIDIUSB library?

orzdk, 2019-09-30:
Thank you for the quick reply. That is good news, then I'm not (nescessarily : ) doing something wrong. 

If you would like a pro micro for testing (no obligations of course) I would be very happy to donate a unit and shipping. Drop me a mail at jes-chr-per@hotmail.com (remove dashes) with a postal address where you can be reached if you are interested. 

Thank you, you can close the issue, I will dig into MIDIUSB to see where the hangup is. 
BR Jesper

orzdk, 2019-09-30:
And yes, you are completely correct, I have tried now, and MIDIUSB lib based solutions seems to work on these particular units. LUFA solutions works fine however, so it must be some board config issue. I will research more.

tttapa, 2019-10-01:
That's a bummer, I would've expected it to work, since it's basically the same chip. You could open an issue on the MIDIUSB GitHub page, or maybe ask SparkFun's support.

If you find a MIDI over USB library that does work on the Pro Micro, feel free to let me know. It's relatively easy to add support for it, all you need is a function to read a USB MIDI packet, and a function to write a USB MIDI packet. (And the necessary USB descriptors, endpoints and setup, of course, but that should happen under the hood.)

orzdk, 2019-10-01:
Me too, it's a bit strange. I have opened issue #57 @ arduino-libraries\MIDIUSB to see if anyone knows anything. Only other library I have been able to get to work is LUFA, and if MIDIUSB is a dead end, I will start to try to isolate the nescessary parts from LUFA, it's quite a big package. 

Thank you kindly for your interest in the matter and for opening up for adding new base libraries, - I will be sure to report my findings upon finalization of research.

orzdk, 2019-10-01:
Hey Pieter. I just closed the issue again on MIDIUSB, the problem seems to be solved !  I burned a fresh bootloader to the unit with an uno, and after that everything worked normally. I suspect that the boards I have, simply was loaded with an ancient, wrong or incorrectly configured bootloader or all of the above. 

Now I will study more and find out how bootloaders relate to USB descriptors, I need to know why it works now. I bet Grumpy Mike would probably love to explain it to me while ripping my a.. in half : P

Lets close the issue, I will not pursue it further for now, and do the actual project instead : ) Be sure to let me know if you want a pro micro for the lab, they're 2.5$ on Ali and i have a big bag of them so it's really nothing. 

Thanks again for your help

orzdk, 2019-10-01:
And on a completely different subject now that im actually up and running, - WOW the pot filter/hysterisis in the library works well. My gritty 1$ pots acts as precision instruments !

closed 2019-10-02 by orzdk

tttapa, 2019-10-02:
Glad to hear you got it working!

The potentiometer filter took some trial and error, but I'm really happy with how they behave now.