tttapa/Control-Surface #41: Led with shift register 74hc595 doesn't work by SPI

kentforth, 2019-10-02:
Hello Pieter!

My board is Arduino Leonardo

I connected a led to shift register SN74HC595N by SPI. It doesn't work. Led blinks fast even if I change a code. It seems it work incorrectly. But if I connect a led in a regular way(digital pins instead of iscp header pins) with a sketch for regular connection it works fine

**Did I make wrong wiring?**

Here is i connected all the stuff:

[image]

Here is my sketch:

`#include <Control_Surface.h> // Include the Control Surface library.

using namespace ExtIO; // Bring the ExtIO pin functions into your sketch

// Instantiate a shift register with the SPI slave select pin as latch pin, most
// significant bit first, and a total of 8 outputs.
SPIShiftRegisterOut<8> sreg = {10, MSBFIRST};

const pin_t ledPin = sreg.pin(1); // first pin of the shift register

void setup() {
  sreg.begin();            // Initialize the shift registers
  pinMode(ledPin, OUTPUT); // You don't even need this line, since
                           // shift registers are always outputs
}

void loop() {
  // Toggle the state of the LED every 1/2 second
  digitalWrite(ledPin, HIGH);
  delay(700);
  digitalWrite(ledPin, LOW);
  delay(700);
}`

tttapa, 2019-10-02:
Master Reset and Output Enable are floating, and the bypass capacitor is missing. See this page: https://www.arduino.cc/en/tutorial/ShiftOut

kentforth, 2019-10-03:
It is strange that author of this article made two different pinouts for a one shift register
pinouts are different. 
[image]
[image]

I don't understand what pinout is the correct one. Can you show me the picture how you connected leds by SPI with shift register when you tested leds?

tttapa, 2019-10-03:
The second one is not a pinout, it's just the schematic, the pins have been rearranged to make routing of the wires easier. The first one is the actual pinout.

kentforth, 2019-10-04:
Here is how I connected Arduino Leonardo to shift register by SPI. It works!
Thank you Pieter!
[image]

closed 2019-10-04 by kentforth

tttapa, 2019-10-04:
I just noticed that the breadboard view on the Arduino page is wrong. **The bypass capacitor should be between Vcc and ground**, as close to the chip as possible. The breadboard (and your diagram above) shows it between ST_CP and ground, which is completely wrong.

kentforth, 2019-10-04:
I checked your suggestion, it works

kentforth, 2019-10-04:
What if I want to connect leds to another shifter register. Can you show me a simple sketch with one led on one shift register and one led on the second shift register?

tttapa, 2019-10-04:
```cpp
#include <Control_Surface.h> // Include the Control Surface library.

using namespace ExtIO; // Bring the ExtIO pin functions into your sketch

// Instantiate a chain of shift registers with the SPI slave select pin as 
// latch pin, most significant bit first, and a total of 16 outputs.
SPIShiftRegisterOut<16> sreg = {SS, MSBFIRST};

const pin_t ledPinA = sreg.pin(0); // first pin of the first shift register
const pin_t ledPinB = sreg.pin(8); // first pin of the second shift register

void setup() {
  sreg.begin();             // Initialize the shift registers
  pinMode(ledPinA, OUTPUT); // You don't even need these lines, since
  pinMode(ledPinB, OUTPUT); // shift registers are always outputs
}

void loop() {
  // Toggle the state of the LEDs every 1/2 second
  digitalWrite(ledPinA, HIGH);
  digitalWrite(ledPinB, LOW);
  delay(500);
  digitalWrite(ledPinA, LOW);
  digitalWrite(ledPinB, HIGH);
  delay(500);
}
```
Change the latch pin from `SS` to pin `10` if you're using a Leonardo.

kentforth, 2019-10-04:
Thank you! I appreciate your work!