tttapa/Control-Surface #42: Program Changer with Banks

PG000, 2019-10-06:
Hi, 
is it possible to make a Program Changer with 4 Buttons in each Bank for sending an individual Program Change command with this library? Because I tried to make the PcBtn bankable but that didn't work. I can't figure out how to accomplish that.
Regards
Paul

tttapa, 2019-10-06:
Hi Paul,

I added the necessary classes and a simple example. The four buttons select four piano voices when the first bank is selected, four organ voices when the second bank is selected, and four guitar/bass voices when the third bank is selected. Did I understand your question correctly?

PG000, 2019-10-06:
Hi tttapa,
thank you very much for making that possible and for the fast response. So thats almost what I was looking for. The only other thing now what I don't understand is, how can I send a serial PC command? I understood how to Set the Midi Output to serial but I didn't understand how to send a PC Command in Serial and how to Set the MIDI Channel for that PC command. Sorry for asking so silly questions.

tttapa, 2019-10-06:
You can change the channel like this:

```cpp
Bankable::ManyAddresses::PCButton<3> pcbuttons[]{
  {bank,
   2, // pin
   {{
     {Acoustic_Grand_Piano, CHANNEL_7},
     {Church_Organ, CHANNEL_4},
     {Acoustic_Guitar_Nylon, CHANNEL_11},
   }}},
  {bank,
   3, // pin
   {{
     {HonkyTonk_Piano, CHANNEL_13},
     {Rock_Organ, CHANNEL_1},
     {Distortion_Guitar, CHANNEL_16},
   }}},
  // ...
};
```

To manually send PC commands, you can do the following:

```cpp
#include <Control_Surface.h>

HardwareSerialMIDI_Interface midi(Serial1, MIDI_BAUD);

using namespace MIDI_PC;

const uint8_t programs[] = { 
  Acoustic_Grand_Piano, 
  Church_Organ, 
  Electric_Bass_Pick, 
};

void setup() {
  midi.begin();
}

void loop() {
  for (uint8_t program : programs) {
    midi.sendPC(CHANNEL_1, program);
    delay(500);
  }
}
```

PG000, 2019-10-07:
Thank you again for the quick response. I don't know if you understood my second question. I want to use the PCButton command to send a PC (0,1,2,3...) on a specific Channel over Serial Midi. I tried to change the code that way. Will that code work ? 

`#include <Control_Surface.h>

HardwareSerialMIDI_Interface midi(Serial, MIDI_BAUD);

Bank<3> bank(4);
IncrementDecrementSelector<3> bankSelector = {bank, {A0, A1}, Wrap::Wrap};

using namespace MIDI_PC;

Bankable::ManyAddresses::PCButton<3> pcbuttons[] = {
  {bank,
   2, // pin
   {{
     {0, CHANNEL_1},
     {1, CHANNEL_1},
     {2, CHANNEL_1},
   }}},
  {bank,
   3, // pin
   {{
     {3, CHANNEL_1},
     {4, CHANNEL_1},
     {5, CHANNEL_1},
   }}},
  {bank,
   4, // pin
   {{
     {6, CHANNEL_1},
     {7, CHANNEL_1},
     {8, CHANNEL_1},
   }}},
  {bank,
   5, // pin
   {{
     {9, CHANNEL_1},
     {10, CHANNEL_1},
     {11, CHANNEL_1},
   }}},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}`

tttapa, 2019-10-07:
That looks fine, at first glance.

PG000, 2019-10-08:
Okay, thank you for your help.

closed 2019-10-08 by PG000