tttapa/Control-Surface #43: Missing define on ESP32

tarTG, 2019-10-11:
I wanted to try your library on an ESP32, but there is a define missing, so i can't compile.

`
/Control-Surface-master/src/Helpers/Exit.cpp: In function 'void CS::fatalErrorExit()':
Control-Surface-master/src/Helpers/Exit.cpp:10:13: 
error: 'LED_BUILTIN' was not declared in this scope
     pinMode(LED_BUILTIN, OUTPUT);
`

Even if i define LED_BUILTIN in my code, it does not compile.

If i select another board, everything works fine.

tttapa, 2019-10-11:
What ESP32 board are you using? Most of them have `LED_BUILTIN` defined, but apparently yours does not.  
I added a check to see if it's defined, so it no longer fails to compile, but keep in mind that there will be no visual feedback when an error occurs, it'll just stop working.  

If this happens, enable `DEBUG_OUT` in `src/Settings/Settings.hpp` and upload the code again. It will then print the error message.

If your board doesn't have a `LED_BUILTIN`, the following warning will be produced:

```
#warning "LED_BUILTIN is not available, so it cannot blink when an error occurs"
```

closed 2019-10-11 by tttapa

tttapa, 2019-10-11:
By the way, if your ESP32 board has a built-in LED, but it's not defined by the Arduino Core, you can open an issue or a pull request on [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32).

For example, this is the `pins_arduino.h` file of a board that is configured correctly:  
https://github.com/espressif/arduino-esp32/blob/79e4339582ca0891a6e6e35220c90db613c0254d/variants/esp32thing/pins_arduino.h#L14-L15

tarTG, 2019-10-11:
Thanks for the fast fix.

commit b0174892 (2019-10-11): Add check for BUILTIN_LED on ESP32
This should fix #43