tttapa/Control-Surface #44: How to use buttons and send Midi CC and Midi Note through multiplexer?

kentforth, 2019-10-15:
I have multiplexer CD74HC4067, I connected 3 buttons to it. 
I didn't find an example how to send midi notes or midi CC through multiplexer
**Can you show me a sketch with one or more buttons?**

I found wiring in your test sketch
CD74HC4067 mux = {
  2,               // input pin
  {3, 4, 5, 6}, // Address pins S0, S1, S2, S3
  // 7, // Optionally, specify the enable pin
};

**should I connect 2 pin on Arduino to pin SIG on multiplexer?**

tttapa, 2019-10-15:
The principle is the same for all MIDI elements, you just pass them a pin number.  
If the pin is a real Arduino pin, the pin number will be `2` or `A0` etc. If it's not a real Arduino pin, you get the pin number from the ExtIO device you're using, e.g. `mux.pin(0)`.

See this example: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d8/d59/1_8First-Output_8ino-example.html. 
And adapt it to use `CCButton` instead of `CCPotentiometer`.

If you specify pin 2 as the first argument to the `CD74HC4067` constructor, then it expects the signal pin to be connected to pin 2.

kentforth, 2019-10-15:
I have downloaded your library again. I opened the sketch "First-Output" when I check it for errors I've got an error:

[image]

**Why it tells me about file with Led if I even didn't use any leds?**

tttapa, 2019-10-15:
Ah, I see, my bad, I added a dependency on the FastLED library, and it seems like I try including it even when it's not installed.

I'll fix it in a moment.

kentforth, 2019-10-15:
Please, leave a message when you fix it. Thanx!

tttapa, 2019-10-15:
In the meantime, you can just install the FastLED library using the library manager, and it should work again.

kentforth, 2019-10-15:
Yes, it works. Just leave a message when you fix this bug. I always want to have new version of library without any bugs:)

tttapa, 2019-10-15:
It should work now.

kentforth, 2019-10-15:
It worx, thank you!

closed 2019-10-19 by kentforth