tttapa/Control-Surface #46: How to change leds' channel used in shift register?

kentforth, 2019-10-19:
Hello Pieter!
I have 3 momentary buttons that connected to multiplexer and send midi notes.
I have 3 leds connected to 3 shift registers

My sketch works fine and leds react to push buttons. The problem is that leds react only on CHANNEL_1, if I change buttons' channel to CHANNEL_2  leds doesn't work.

**How can I change leds' channel that react on midi notes?**

Here is a piece of code from my sketch:

```
NoteButton button[] = {
  {mux.pin(0),{note(C, 4), CHANNEL_1}},
  {mux.pin(7),{note(D, 4), CHANNEL_1}},
  {mux.pin(15),{note(E, 4), CHANNEL_1}}
};
  
// Create an array of LEDs that listen to MIDI Note messages, turning on and off
// the LEDs connected to the eight input pins of the shift register
MIDINoteLED leds[] = {
  {sreg.pin(1), note(C, 4)}, // LED pin, address (note number, channel, cable)
  {sreg.pin(9), note(D, 4)},
  {sreg.pin(17), note(E, 4)}
};
```

tttapa, 2019-10-19:
The address is always an object of type `MIDICNChannelAddress`. The constructor takes three parameters: the MIDI address (note/controller number), the MIDI channel, and the MIDI cable number.

The latter two have default values, if you don't specify them, the channel is 1, and the cable number is 0.  
You can specify those optional parameters, but you have to add braces to group them:
```cpp
MIDICNChannelAddress address1 = note(C, 4); // default channel, default cable
MIDICNChannelAddress address2 = {note(C, 4), CHANNEL_1}; // default cable
MIDICNChannelAddress address3 = {note(C, 4), CHANNEL_1, 0}; // no defaults
```

<https://tttapa.github.io/Control-Surface/Doc/Doxygen/d3/d10/classMIDICNChannelAddress.html#af01126768cafcda0d97f3297c65222c7>

```cpp
NoteButton button[] = {
  {mux.pin(0),{note(C, 4), CHANNEL_2}},
  {mux.pin(7),{note(D, 4), CHANNEL_2}},
  {mux.pin(15),{note(E, 4), CHANNEL_2}}
};

MIDINoteLED leds[] = {
  {sreg.pin(1), {note(C, 4), CHANNEL_2}}, // LED pin, address (note number, channel, cable)
  {sreg.pin(9), {note(D, 4), CHANNEL_2}},
  {sreg.pin(17), {note(E, 4), CHANNEL_2}}
};
```

kentforth, 2019-10-19:
It works very well thank you!

closed 2019-10-19 by kentforth