tttapa/Control-Surface #47: How to make leds react on MidiCC instead of notes?

kentforth, 2019-10-19:
I have 3 shift registers and 3 leds. 
You have an example to use leds that reacts on MidiNote

**Is it possible to make leds react on MidiCC instead? How can I do that?**

MIDINoteLED leds[] = {
  {sreg.pin(1), note(C, 4)}, // LED pin, address (note number, channel, cable)
  {sreg.pin(9), note(D, 4)},
  {sreg.pin(17), note(E, 4)}
};

tttapa, 2019-10-19:
Yes, use `CCValueLED` instead: https://tttapa.github.io/Control-Surface/Doc/Doxygen/da/d34/classCCValueLED.html#a4d4ed28508873e3f2d5a557dc3326a22

(BTW, MIDINoteLED is deprecated, prefer using `NoteValueLED` instead, it's consistent with the rest of the API.)

kentforth, 2019-10-19:
Can't see examples there

**Can you show me one example?**

tttapa, 2019-10-19:
Just replace `MIDINoteLED` in your code above with `CCValueLED`.

kentforth, 2019-10-19:
**I mean How to properly pass arguments with midi cc to leds?**

I tried to do it like that, but it doesn't work:

```
CCValueLED leds[] = {
  {sreg.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {sreg.pin(9), {MIDI_CC::Pan, CHANNEL_2}},
  {sreg.pin(17),{MIDI_CC::Balance, CHANNEL_3}}
};
```

tttapa, 2019-10-19:
Do you have the latest version?

I'll try it in a couple of minutes, but I think it should work on the latest version.

kentforth, 2019-10-19:
I downloaded your library again.
I have this code:

  
```
NoteButton button[] = {
  {mux.pin(0),{MIDI_CC::Channel_Volume, CHANNEL_1}},
  {mux.pin(7),{MIDI_CC::Pan, CHANNEL_2}},
  {mux.pin(15),{MIDI_CC::Balance, CHANNEL_3}}
};

CCValueLED leds[] = {
  {sreg.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {sreg.pin(9), {MIDI_CC::Pan, CHANNEL_2}},
  {sreg.pin(17),{MIDI_CC::Balance, CHANNEL_3}}
};
```

Leds doesn't react and I have used Midi OX to display incoming messages. What I have now:
[image]

It seems the library defines Midi:CC commands as regular notes but different ones

tttapa, 2019-10-19:
I don't understand. You're sending CC addresses using a `NoteButton`. Try a `CCButton` instead.

kentforth, 2019-10-19:
Oops, my bad:)
Thank you for your help! It works

closed 2019-10-19 by kentforth

tttapa, 2019-10-19:
Glad to hear!