tttapa/Control-Surface #49: Toggle buttons and leds and send midi cc

kentforth, 2019-10-22:
Hello Pieter!
I have buttons  in multiplexer 4067 and leds in shift registers. 

When I press a button midi cc with value should be sent and led will light up, when I press the same button again midi cc with another value should be sent and led will go out.
Buttons should be like toggles.
It's like when you press a a button on a footswitch to change preset on midi controller in some guitar VST like Guitar Rig or Bias FX.

[image]

**I saw CCButtonLatched class, is it what I should use for my task?**
I didn't find an example with toggle buttons and leds
**Can you write me an example how to use buttons as toggles with leds?**

tttapa, 2019-10-22:
Something like this?

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

CD74HC4051 mux = {
  2,        // Input pin
  {3, 4, 5} // Address pins S0, S1, S2
};

SPIShiftRegisterOut<8> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

CCButtonLatched buttons[] = {
  {mux.pin(0), 0},
  {mux.pin(1), 1},
  {mux.pin(2), 2},
  {mux.pin(3), 3},
  {mux.pin(4), 4},
  {mux.pin(5), 5},
  {mux.pin(6), 6},
  {mux.pin(7), 7},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
  uint8_t i = 0;
  for (CCButtonLatched &button : buttons)
    sreg.digitalWrite(i++, button.getState());
}
```

kentforth, 2019-10-22:
Yes, it works! 
**But how can I change Midi CC number of particular buttons on multiplexer?**

tttapa, 2019-10-22:
The second argument to the `CCButtonLatched` constructor is the address, as usual. Right now they are 0, 1, 2 etc.

kentforth, 2019-10-22:
Thank you very much!

closed 2019-10-22 by kentforth