tttapa/Control-Surface #50: How to use rgb leds with common anode?

kentforth, 2019-10-23:
Hello Pieter!
I have one shift register and 3 RGB Leds.
I have 3 buttons in multiplexer
When I press first button red color of led should light up. WHen I press it again a led light should go out.
Second button - green color of second led
Third button - blue color of third led
Usually RGB leds have common cathode, but my rgb leds have common anode.
**How should I wire rgb led to shift register in this case?**

**Can you give me a simple example of buttons and rgb leds for my case?**

tttapa, 2019-10-23:
[image]

The code is similar to the code you used before, just invert the state:

```cpp
sreg.digitalWrite(i++, !button.getState());
```

kentforth, 2019-10-23:
**How can I set particular color of any rgb leds?**
For example: I want to press a button and led for that button should have 135,255,241 color

tttapa, 2019-10-23:
That's not possible, normal shift registers are on/off only.

kentforth, 2019-10-23:
I mean not setting color during usage of arduino but before uploading the sketch. I just want have another color of one rgb led instead of blue or green or red

kentforth, 2019-10-23:
**Can you give me an example of one button on multiplexer that toggle a led on shift register with color 
135,255,241?**

tttapa, 2019-10-23:
No, you can't do that. Shift registers can only turn on or off an LED, they can't dim it.

kentforth, 2019-10-23:
So, rgb led on shift registers can have only red or blue or green color?
If so, where Can I declare that color?

tttapa, 2019-10-23:
No, they can display red, green, blue, yellow, cyan, magenta and black.  
RGB LEDs are just three LEDs close together, you don't set the color, you decide which of the three LEDs to turn on.

For example, to get magenta on the first RGB LED:
```cpp
sreg.digitalWrite(0, HIGH);  // red on
sreg.digitalWrite(1, LOW);   // green off
sreg.digitalWrite(2, HIGH);  // blue on
```

kentforth, 2019-10-23:
sreg.digitalWrite(0, HIGH);
0 - is the pin on multiplexer?

tttapa, 2019-10-23:
Yes.

It's equivalent to `digitalWrite(sreg.pin(0), HIGH)` (but faster).

kentforth, 2019-10-24:
It's hard to understand how to work with rgb leds=(
I connected first button to multiplexer pin 0, second - 1 third button - 2 pin
I connected first RGB Led to shift register pin 0, second - 1, third to 2 pin
Second RGB LED - 3,4,5
Third RGB LEd - 6,7,8

When I press first button I want first RGB LED light up with red color
Second button - second led with green color
Third button - third led with blue color

My sketch is wrong because when I connect arduino to PC led light up with red color even I didn't press a button
**Can you help me with sketch?**

```
#include <Control_Surface.h>  // Include the library

using namespace ExtIO;

USBMIDI_Interface midi; // Instantiate a MIDI Interface to use

// Instantiate a multiplexer
 CD74HC4067 mux = {
  2,               // input pin
  {3, 4, 5, 6},    // Address pins S0, S1, S2, S3
  // 7,             // Optionally, specify the enable pin
};

// Instantiate a shift register as output for the LEDs
SPIShiftRegisterOut<24> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

// Instantiate a CCButtonLatched object
CCButtonLatched buttons[] = {
  {mux.pin(0), 5},
  {mux.pin(1), 6},
  {mux.pin(2), 7},
  {mux.pin(3), 8},
  {mux.pin(4), 9}, 
  {mux.pin(5), 10},
  {mux.pin(6), 11},
  {mux.pin(7), 12},  
};

void setup() {
  Control_Surface.begin();   // Initialize Control Surface
}

void loop() {
  Control_Surface.loop();   // Update the Control Surface
      
      if (buttons[0].getState() == Button::Falling) {
       sreg.digitalWrite(0, LOW);  // red on
      sreg.digitalWrite(1, HIGH);   // green off
      sreg.digitalWrite(2, HIGH);  // blue on
      }
      else if (buttons[1].getState() == Button::Falling) {
       sreg.digitalWrite(3, HIGH);  // red on
      sreg.digitalWrite(4, LOW);   // green off
      sreg.digitalWrite(5, HIGH);  // blue on
      }
      else if (buttons[2].getState() == Button::Falling) {
       sreg.digitalWrite(3, HIGH);  // red on
      sreg.digitalWrite(4, HIGH);   // green off
      sreg.digitalWrite(5, LOW);  // blue on
      }
}
```

tttapa, 2019-10-25:
I think you have to add code to turn off the LEDs. And just use separate if statements, not if-else-if.

tttapa, 2019-10-25:
Oh, I see, you're comparing the state to `Button::Falling`, but the latched buttons return `true` or `false`.

kentforth, 2019-10-25:
so should I change `Button::Falling` to `Button::Rising`?

tttapa, 2019-10-25:
```cpp
#include <Control_Surface.h>  // Include the library

using namespace ExtIO;

USBMIDI_Interface midi; // Instantiate a MIDI Interface to use

// Instantiate a multiplexer
 CD74HC4067 mux = {
  2,               // input pin
  {3, 4, 5, 6},    // Address pins S0, S1, S2, S3
  // 7,            // Optionally, specify the enable pin
};

// Instantiate a shift register as output for the LEDs
SPIShiftRegisterOut<24> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

// Instantiate an array of CCButtonLatched objects
CCButtonLatched buttons[] = {
  {mux.pin(0), 5},
  {mux.pin(1), 6},
  {mux.pin(2), 7},
  {mux.pin(3), 8},
  {mux.pin(4), 9}, 
  {mux.pin(5), 10},
  {mux.pin(6), 11},
  {mux.pin(7), 12},  
};

constexpr uint8_t COMMON_ANODE = 1;
constexpr uint8_t COMMON_CATHODE = 0;

constexpr uint8_t red = 1;
constexpr uint8_t green = 2;
constexpr uint8_t blue = 4;

// Change this if necessary
constexpr uint8_t mode = COMMON_ANODE;

// Choose colors here
constexpr uint8_t colors[] = {
  blue,   // first LED will be blue
  green,  // second LED will be green
  red,
  red | blue,          // magenta
  green | blue,        // cyan
  red | green,         // yellow
  red | green | blue,  // white
  red,
};

void setup() {
  Control_Surface.begin();   // Initialize Control Surface
  for (pin_t pin : sreg.pins())
    digitalWrite(pin, mode);
}

void loop() {
  Control_Surface.loop();   // Update the Control Surface

  uint8_t i = 0;
  for (auto &button : buttons) {
    if (button.getButtonState() == Button::Falling && button.getState() == true) {
      sreg.digitalWrite(i * 3 + 0, mode ^ !!(colors[i] & red));
      sreg.digitalWrite(i * 3 + 1, mode ^ !!(colors[i] & green));
      sreg.digitalWrite(i * 3 + 2, mode ^ !!(colors[i] & blue));
    } else if (button.getButtonState() == Button::Falling && button.getState() == false) {
      sreg.digitalWrite(i * 3 + 0, mode);
      sreg.digitalWrite(i * 3 + 1, mode);
      sreg.digitalWrite(i * 3 + 2, mode);
    }
    ++i;
  }
}
```

kentforth, 2019-10-25:
I have an error
[image]

tttapa, 2019-10-25:
Did you update to the latest version?

tttapa, 2019-10-25:
> so should I change `Button::Falling` to `Button::Rising`?

No, `buttons[0].getState()` returns `true` or `false`. You should compare it to `true` or `false`, not  `Button::Falling` or `Button::Rising`.

kentforth, 2019-10-25:
I connected one led to pins 0,1,2 (R,G,B) of shift register just to try, when I press first button nothing happens. What should I change in code here?

tttapa, 2019-10-25:
I don't know, I have it working on an Arduino right in front of me. Did you change anything to the code?

kentforth, 2019-10-25:
constexpr uint8_t red = 0;
constexpr uint8_t green = 2;
constexpr uint8_t blue = 4;

**What are these pins? is it pins of first rgb led on shjift register?**

tttapa, 2019-10-25:
These are not pins, they are color constants, you shouldn't have to change them.

kentforth, 2019-10-25:
I connected all 3 RGB  leds. The Second Led work perfectly and green color light up when button is pressed, but the first and third buttons doesn't work. Do you have any idea?

kentforth, 2019-10-25:
Buttons work perfectly because I tested them in MIDI OX. Buttons sends midi cc as usual

tttapa, 2019-10-25:
What board are you using?

kentforth, 2019-10-25:
Arduino Leonardo

tttapa, 2019-10-25:
How did you wire the shift registers and the LEDs? Do the other LEDs light up at all, or are they all off?

kentforth, 2019-10-25:
I will make wiring in Photoshop just after a couple of minutes

kentforth, 2019-10-25:
Led 1 :
	R -> Q0
	G -> Q1
	B -> Q2
	Anode -> Vcc

Led 2:
	R -> Q3
	G -> Q4
	B -> Q5
	Anode -> Vcc

Led 3:
	R -> Q6
	G -> Q7
	B -> Q0 of second shift register
	Anode -> Vcc

[image]

kentforth, 2019-10-25:
Initially all leds are off. when I press second button a second led is On with green color, others doesn't work

kentforth, 2019-10-25:
I think may be something wrong with these constants:

```
constexpr uint8_t red = 0;
constexpr uint8_t green = 2;
constexpr uint8_t blue = 4;

// Change this if necessary
constexpr uint8_t mode = COMMON_ANODE;

// Choose colors here
constexpr uint8_t colors[] = {
  blue,   // first LED will be blue
  green,  // second LED will be green
  red,
  red | blue,          // magenta
  green | blue,        // cyan
  red | green,         // yellow
  red | green | blue,  // white
  red,
};
```

you set blue as first LED , but I connected red as first pin on first shift register. 
**Does it matter?**

kentforth, 2019-10-25:
Sorry Pieter! it is my bad=( my jumpers were broken. I had bad wiring. Now all three leds are working but with inverted colors and  I changed the code:

```
constexpr uint8_t colors[] = {
  red,   // first LED will be blue
  green,  // second LED will be green
  blue,
  red | blue,          // magenta
  green | blue,        // cyan
  red | green,         // yellow
  red | green | blue,  // white
  red,
};
```

kentforth, 2019-10-25:
Thank you for your help! I appreciate your work!

closed 2019-10-25 by kentforth

tttapa, 2019-10-25:
Glad to hear you got it working!

To fix the color names, you can just swap the pin offset (swap the 2 and the 0):

```cpp
      sreg.digitalWrite(i * 3 + 2, mode ^ !!(colors[i] & red));
      sreg.digitalWrite(i * 3 + 1, mode ^ !!(colors[i] & green));
      sreg.digitalWrite(i * 3 + 0, mode ^ !!(colors[i] & blue));
```

kentforth, 2019-11-13:
Hello Pieter!
I have this code :
```
#include <Control_Surface.h>  // Include the library

using namespace ExtIO;

USBMIDI_Interface midi; // Instantiate a MIDI Interface to use

// Instantiate a multiplexer
 CD74HC4067 mux = {
  2,               // input pin
  {3, 4, 5, 6},    // Address pins S0, S1, S2, S3
  // 7,            // Optionally, specify the enable pin
};

// Instantiate a shift register as output for the LEDs
SPIShiftRegisterOut<24> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

// Instantiate an array of CCButtonLatched objects
CCButtonLatched buttons[] = {
  {mux.pin(0), 5},
  {mux.pin(1), 6},
  {mux.pin(2), 7},
  {mux.pin(3), 8},
  {mux.pin(4), 9}, 
  {mux.pin(5), 10},
  {mux.pin(6), 11},
  {mux.pin(7), 12},  
};

constexpr uint8_t COMMON_ANODE = 1;
constexpr uint8_t COMMON_CATHODE = 0;

constexpr uint8_t red = 1;
constexpr uint8_t green = 2;
constexpr uint8_t blue = 4;

// Change this if necessary
constexpr uint8_t mode = COMMON_ANODE;

// Choose colors here
constexpr uint8_t colors[] = {
  red,   // first LED will be red
  green,  // second LED will be green
  blue,
  red | blue,          // magenta
  green | blue,        // cyan
  red | green,         // yellow
  red | green | blue,  // white
  red,
};

void setup() {
  Control_Surface.begin();   // Initialize Control Surface
  for (pin_t pin : sreg.pins())
    digitalWrite(pin, mode);
}

void loop() {
  Control_Surface.loop();   // Update the Control Surface

  uint8_t i = 0;
  for (auto &button : buttons) {
    if (button.getButtonState() == Button::Falling && button.getState() == true) {
      sreg.digitalWrite(i * 3 + 0, mode ^ !!(colors[i] & red));
      sreg.digitalWrite(i * 3 + 1, mode ^ !!(colors[i] & green));
      sreg.digitalWrite(i * 3 + 2, mode ^ !!(colors[i] & blue));
    } else if (button.getButtonState() == Button::Falling && button.getState() == false) {
      sreg.digitalWrite(i * 3 + 0, mode);
      sreg.digitalWrite(i * 3 + 1, mode);
      sreg.digitalWrite(i * 3 + 2, mode);
    }
    ++i;
  }
}
```

But I want  to change color of third button from blue to magenta
**How can I do this?**

I tried to change from :
`sreg.digitalWrite(i * 3 + 2, mode ^ !!(colors[i] & blue));`

to : 
`sreg.digitalWrite(i * 3 + 2, mode ^ !!(colors[i] & red | blue));`

but first and second buttons' colors are changed and the third one doesn't work at all. I just need to change color for third button

tttapa, 2019-11-13:
Don't change the digitalWrite lines, change the third entry in the `colors` array.

tttapa, 2019-11-13:
The bitwise or operator `|` mixes two colors, bitwise and `&` selects one of the three colors, and `!!` checks if it is turned on or not.

For example, magenta is `red | blue = 1 | 2 = 0b011`.  Let's say `uint8_t color = red | blue;`  
Then `color & red = 0b011 & 1 = 0b001` to select only the red part. Finally, `!! 0b001 = 1`, so the red LED should be on.  
For blue, `color & blue = 0b011 & 2 = 0b010` and `!! 0b010 = 1`, so the blue LED should be on as well.  
For green, `color & green = 0b011 & 4 = 0b000` and `!! 0b000 = 0`, so the green LED should be off.

The state of the LEDs is then XORed with the mode, to invert it if the mode is common anode.

kentforth, 2019-11-13:
What if I want to have 3 buttons that have red, green, blue color and add one more button that have cyan color. 
**Where should I change the code?**

tttapa, 2019-11-13:
```cpp
constexpr uint8_t colors[] = {
  red,                 // first LED will be red
  green,               // second LED will be green
  blue,                // third LED will be blue
  green | blue,        // Fourth LED will be cyan
  red | blue,          // Fifth LED will be magenta
  red | green,         // Sixth LED will be yellow
  red | green | blue,  // Seventh LED will be white
  red,                 // Eighth LED will be red
};
```

kentforth, 2019-11-13:
In this case button 4 is blue color

tttapa, 2019-11-13:
It works for me. Did you check your wiring?

tttapa, 2019-11-13:
> Should I add one more line in scope with digitalWrite?

No.

kentforth, 2019-11-14:
I checked wiring. It is ok
Do you have any ideas?

<a href="https://imgur.com/KSNcOaH" target="_blank"><img src="https://imgur.com/SNfeO0z" 
alt="VIdeo" width="240" height="180" border="10" /></a>

tttapa, 2019-11-14:
I used the code I posted [here](https://github.com/tttapa/Control-Surface/issues/50#issuecomment-546410067) and it works just fine.

Maybe you blew up one of the LEDs or shift registers. You don't seem to be using any current limiting resistors, you need those.

kentforth, 2019-11-15:
I used the same code the you left here. 3 colors works fine as expected. 1  button  - blue, 2 - green, 3 - red. But the fourth button has blue color
I think this `red | blue` magenta color doesn;t work.
I also connected resistors to leds but I think they doesn't change leds' behavior, only it's color brightness

tttapa, 2019-11-15:
It won't change the behavior, but maybe you blew up something by not using any resistors.  
Try using red | blue for the first LED, for example.

You can convince yourself that the code is correct by printing the values you're sending to the shift registers.

kentforth, 2019-11-15:
I changed first led to red | blue and it works:) but it doesn't work for 4 led O_o

tttapa, 2019-11-15:
Can you test the LED without the shift register? (With just a resistor to ground.)

What pins of the shift register is it connected to?

kentforth, 2019-11-15:
4 led is connected to second shift register( Red -> Q1, Green -> Q2, Blue -> Q3)

I don't know how to test a led.  Do you mean to find some random sketch for a button with rgb led?

tttapa, 2019-11-15:
[image]

V+ is the Arduino's 5V pin, V- is the Arduino's ground pin.

kentforth, 2019-11-16:
What  I -> means in this schema?

kentforth, 2019-11-16:
Should I connect rgb led for testing like that?

[image]

tttapa, 2019-11-16:
I is the current, the arrow indicates the direction of conventional current flow.

The image looks fine, except that the black wire should be connected to the R pin.

kentforth, 2019-11-16:
I'm an idiot=(
The problem was wiring. These chinese jumpers are awfull. I made my own jumpers from ethernet cable and all leds works fine!
Thank you Pieter for helping me!

tttapa, 2019-11-16:
Glad to hear you got it working!