tttapa/Control-Surface #52: How can I change button's note with master button?

AngleCreDible, 2019-10-26:
I tried to semitone the notes assigned to all buttons while pressing the 'master button'.
```
...
#define MASTERBTN      42
boolean isMaster = false;
...
char noteNames[] = {C, D, E, F, G, A, B};
int noteNum = 3;
...
NoteButton buttons[] = {
  {43, {note(noteNames[0], noteNum), CHANNEL_1}},
  ...
  {33, {note(noteNames[0], noteNum + 2), CHANNEL_1}}
};
...
void loop() {
  if(digitalRead(MASTERBTN) == LOW) {
    if(!isMaster) {
      isMaster = true;
      noteNames[0] = C;
      noteNames[1] = Db;
      noteNames[2] = Eb;
      noteNames[3] = F;
      noteNames[4] = Gb;
      noteNames[5] = Ab;
      noteNames[6] = Bb;
      Serial.println("here we are");
    }
  } else {
    if(isMaster) {
      isMaster = false;
      noteNames[0] = C;
      noteNames[1] = D;
      noteNames[2] = E;
      noteNames[3] = F;
      noteNames[4] = G;
      noteNames[5] = A;
      noteNames[6] = B;
      Serial.println("now here");
    }
  }
  Control_Surface.loop();
  FastLED.show();
}
```
I could check the serial messages, but the log and MIDI signals did not output the modified note.
How can I change button's note with master button?

tttapa, 2019-10-26:
Changing `noteNames` doesn't change the addresses of the buttons.

I think what you need is the [`ManyAddresses::NoteButton`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/d2/d33/classBankable_1_1ManyAddresses_1_1NoteButton.html).  
That's the best way to do it if you need arbitrary addresses.

If you just want to transpose everything by the same amount of semitones, it's best to use the [`Bankable::NoteButton`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/de/d36/classBankable_1_1NoteButton.html) class combined with a [`Transposer`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/da/db8/classTransposer.html).

AngleCreDible, 2019-10-27:
Can you show me some examples?

AngleCreDible, 2019-10-27:
In addition, Is it pottible to make when you press button 1, you increase one octave, when you press button 2, you go one octave down, when you press button 3, you can change the major order (F-C-G-D-A-E-B to Gb-Db-Ab-Eb-Bb-Fb-C), when the button 4 is pressed, return the majors in order?
(sorry this is translated comment, so it can be strange.)

tttapa, 2019-10-27:
Sure, basically, you create a custom class that has a `lock`, `unlock` and `getActiveAddress` method.

Lock and unlock ensure that the note number doesn't change while the button is pressed (otherwise the note off event will be sent to a different address, leaving the original note playing forever).

GetActiveAddress returns the right address, in your case, the address of the right note (F or Gb) plus or minus one octave.

`ManyAddresses` is a class that handles a list of alternative addresses, and `BankableMIDIOutput` is a class that adds an offset to a given address, so both are combined:

https://github.com/tttapa/Control-Surface/blob/8ebcc4dcfecbc6c002e3e57be6c85a85973fe2d3/examples/Other/GitHub-issues/Note-ManyAddresses-Transposer/ManyAddressBankNoteButton.hpp

Then you just pass it as the first template argument to the `Bankable::MIDIButton` class.

It looks scary, but it's all just glue between existing classes. You probably don't have to change that file.

What you do have to change is the note numbers etc. in the main sketch:  
https://github.com/tttapa/Control-Surface/blob/8ebcc4dcfecbc6c002e3e57be6c85a85973fe2d3/examples/Other/GitHub-issues/Note-ManyAddresses-Transposer/Note-ManyAddresses-Transposer.ino

AngleCreDible, 2019-10-27:
That looks really nice, I'll try it.
I really love you and this library!

tttapa, 2019-10-27:
Oh, by the way, if you're not seeing any MIDI output, that could be because the sketch uses a `USBDebugMIDI_Interface`, it prints the MIDI message in a readable format to the serial monitor. You can change it to a `USBMIDI_Interface` if you want.

AngleCreDible, 2019-10-27:
Can I ask one more question?
When there is a Bank <N>, is it possible to press the button to go to the last bank and press the button again to return to the previous bank?

tttapa, 2019-10-27:
Not sure what you mean.

The `Bank` class has a `select` method that you can call directly: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/db/dbd/classBank.html#ab4b91b7f4363e5e497b34e17a303bdbb>

There are some included `Selector`s that can manage a `Bank`, based on the input from push buttons or rotary encoders etc. <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d0/d0c/group__Selectors.html>

AngleCreDible, 2019-11-02:
It works very nice, thanks!

closed 2019-11-02 by AngleCreDible