tttapa/Control-Surface #53: OLED display + midi clock

Sixonlon, 2019-10-31:
I see that you already have implemented OLED display possibilities but it is for spi displays. Will it be possible to have the same for I2C ? 

Also, I don't find a way to use midi clock to display the bpm from my DAW with,for example, a led. Is it in your plans ?

tttapa, 2019-10-31:
Yes, the OLED code can be used with I2C displays as well, you can find how to do it in the Adafruit_SSD1306 library examples, there should be no difference in the Control Surface code. Do keep in mind that the I2C variant is much slower than SPI. You can increase the I2C bus speed on most boards, but it probably still won't be as fast as SPI.

What DAW are you using? In what format is it sending the BPM information?

Sixonlon, 2019-10-31:
Ok ty !

And I'm currently using Ableton live

closed 2021-02-10 by tttapa