tttapa/Control-Surface #55: Send MIDI Note value through a potentiometer.

Skipazz, 2019-11-07:
Hi!

Hope you're doing well.
I am currently in the beginning process of building a MIDI controller which works in a slightly unusual way and I am not too sure where to head in terms of writing the code for one specific action.

To be precise, I would like to use the variable resistance of a potentiometer to assign specific note values to each knob and have it in a way such as:

the potentiometer is all the way to the left at 0 = note off
as it turns clockwise = note on (a specific note for example middle c) and for each incremental change in the position of the knob have it reflect into an increase of midi velocity with the max being at 127.

How is it possible to do this?

Thank you so much for your time in advance,

Kindly

Gio

tttapa, 2019-11-07:
When should it send the note events? Should polyphone aftertouch be used, or is sending multiple events fine? Or is sending the note controlled by a separate button?

Skipazz, 2019-11-07:
The basic idea is that I would like to completely replace any push buttons in the set up and only use pan pots, so the note should send when position is > 0 and stop sending only when the position is returned to the initial point at 0

I would like each knob to control a different note, so for instance knob 1 = C, Knob 2 = C#, etc..

tttapa, 2019-11-07:
```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

FilteredAnalog<7> analog = A0;

using namespace MIDI_Notes;
const MIDICNChannelAddress midinote = {note(C, 4), CHANNEL_1};

bool noteOnSent = false;

void setup() {
  Control_Surface.begin();
}

void loop() {
  if (analog.update()) {
    uint8_t newValue = analog.getValue();
    if (newValue > 0) {
      if (noteOnSent) {
        midi.sendKP(midinote, newValue); // MIDI Key Pressure
      } else {
        midi.sendNoteOn(midinote, newValue);
        noteOnSent = true;
      }
    } else {
      midi.sendNoteOff(midinote, 0x7F);
      noteOnSent = false;
    }
  }
}
```

tttapa, 2019-11-07:
If your synthesizer doesn't support key pressure / aftertouch, you'll have to send new notes every time. On the synth I tried it with, it sounds absolutely horrible, because it triggers a new note each time, but YMMV.

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

FilteredAnalog<7> analog = A0;

using namespace MIDI_Notes;
const MIDICNChannelAddress midinote = {note(C, 4), CHANNEL_1};

void setup() {
  Control_Surface.begin();
}

void loop() {
  if (analog.update())
    midi.sendNoteOn(midinote, analog.getValue());
}
```

Skipazz, 2019-11-07:
Thank you so much for your quick reply and help. I have tried both codes and whilst they both work, I am not sure this is the result I'm after. Both sketches trigger the note quite far up the potentiometer (I'm using a 10k linear pot). But once the note has been triggered, it doesn't sustain which is what I need. Would it help at all if I sent you a picture of the very basic hardware set up I have right now for testing? it only consists of one single pot plugged into A0 and then simply 5V and ground into the arduino

tttapa, 2019-11-07:
Sustain is handled by the synthesizer, not the controller.

Skipazz, 2019-11-07:
Thank you so much!
In case I wanted to add more pots, what is the correct syntax? I apologise but still quite haven't gotten my head round this library!

tttapa, 2019-11-07:
```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

class NotePot : public MIDIOutputElementPotentiometer {
  public:
    NotePot(pin_t analogPin, MIDICNChannelAddress midinote)
      : analog(analogPin), midinote(midinote) {}

    void begin() override {}
    void update() override {
      if (analog.update()) {
        uint8_t newValue = analog.getValue();
        if (newValue > 0) {
          if (noteOnSent) {
            midi.sendKP(midinote, newValue); // MIDI Key Pressure
          } else {
            midi.sendNoteOn(midinote, newValue);
            noteOnSent = true;
          }
        } else {
          midi.sendNoteOff(midinote, 0x7F);
          noteOnSent = false;
        }
      }
    }

    void map(MappingFunction fn) { analog.map(fn); }
    void invert() { analog.invert(); }

  private:
    FilteredAnalog<7> analog;
    MIDICNChannelAddress midinote;
    bool noteOnSent = false;
};

using namespace MIDI_Notes;
NotePot notepots[] = {
  {A0, {note(C, 4), CHANNEL_1}},
  {A1, {note(Db, 4), CHANNEL_1}},
  {A2, {note(D, 4), CHANNEL_1}},
  {A3, {note(Eb, 4), CHANNEL_1}},
  // Etc.
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

Skipazz, 2019-11-07:
Thank you so very much! 
If I wanted to dive deeper into this library to start learning it, where would you recommend me to begin from? I’m very new to coding and I’m still learning and have a long road ahead of me so any tips are welcome and I thank you again for your time and patience!

tttapa, 2019-11-07:
There's a basic [**Getting Started page**](https://tttapa.github.io/Control-Surface/Doc/Doxygen/dd/dcc/md_Getting-Started.html). To learn about the many features of the library, you can look at the [**Modules page**](https://tttapa.github.io/Control-Surface/Doc/Doxygen/modules.html) of the documentation.  

If you're new to coding, the detailed documentation is probably hard to read, and contains way too much information for basic use cases.  
The [**examples**](https://tttapa.github.io/Control-Surface/Doc/Doxygen/examples.html) are the best place to start in that case.  
Writing examples is a lot of work, so there are probably many features without a concrete example. Don't hesitate to open an issue, I can write an example for the feature if it's missing.

Skipazz, 2019-11-07:
Thanks a lot! I have been reading through the examples now and will try implementing the sketches! As soon as I get a hang of it I might even try writing my own examples for those features that don’t have one uploaded yet just to learn.

Skipazz, 2019-11-08:
Hey Pieter!

Thanks again for all your help. I was going to ask you, as you might know better than me, if you had any quick tips or pointers to solve this issue.

The code that I was asking you help with is to be implemented in a midi controller which is light sensitive. The idea is that I will have a bunch of photoresistors, each to its own analog pin of an arduino and map them to send specific midi note values, so instead of pushing a key to play the note, you would cover the photo cell in a way such as with light on = no sound is emitted, as the cell begins to be covered the midi signal is sent up until there is no more light passing through to the photoresistsor which will equal velocity max for the midi note.

I have tried using what we have in our labs at college but the resistors seem to be way too sensitive to light, to the point that the minimal light would trigger them to send note on. (I still have not figured out a way to invert so for now i have light off = note off and light on = note on but with not sensitivity in the velocity unfortunately).
DO you know by any chance which photoresitor would work well for this project? what kind of sensitivity and how much resistance?

I'll attach a video showing a basic prototype with two cells connected to the Operator synth in ableton live 10.

https://drive.google.com/drive/folders/1GVGEOx62up4gRDhBd3ERLhFQTH_ExfYm?usp=sharing

tttapa, 2019-11-08:
The sensitivity of the photoresistors doesn't really matter, you can just change the biasing current by changing the resistor value of the second resistor.

The first thing to try is to use the simple analogReadSerial example and check how the sensors react. Then you can use the map function to map it to your likings.

Skipazz, 2019-11-09:
Thanks! Do you have any documentation on the sendKP function ?

tttapa, 2019-11-09:
It just sends a MIDI Key Pressure event.  
From the MIDI Standard:

> ## Aftertouch
> Two types of Aftertouch messages are available: one that affects an entire MIDI channel and one that affects each individual note played. They are differentiated by their status byte. In either case, the Aftertouch value is determined by horizontally moving the key (front-to-rear or left-to-right), or by pressing down on the key after it "bottoms out". Devices such as wind controllers can send Aftertouch from increasing breath pressure after the initial attack. The type of tone modification created by the Aftertouch is determined by the receiver. Aftertouch may be assigned to affect volume, timbre, vibrato, etc.
>
> If a "Channel Pressure" (Dn, 0vvvvvvv) message is sent, then the Aftertouch will affect all notes playing in that channel.
>
> If a "Polyphonic Key Pressure" (An, 0kkkkkkk, 0vvvvvvv) message is sent discrete Aftertouch is applied to each note (0kkkkkkk) individually.

The `sendKP` takes just two parameters, the MIDI address and the pressure value ranging from 0 to 127:

```cpp
    /// Send a MIDI Key Pressure event.
    void sendKP(MIDICNChannelAddress address, uint8_t pressure);
```

For example: 

```cpp
USBMIDI_Interface midi;
midi.sendKP({note(C, 4), CHANNEL_3, 10}, 0x7F);
```
Sends a MIDI Key Pressure event for note C4 (middle C) on MIDI channel 3, on USB MIDI Cable 10 (the 11th cable) and with a pressure value (cfr. velocity) of 127 or 0x7F.  

As with all addresses in the Control Surface library, the cable number and channel are optional:

```cpp
midi.sendKP({note(C, 4), CHANNEL_3}, 0x7F); // Cable 0 (first and default cable)
midi.sendKP(note(C, 4), 0x7F); // Channel 1, Cable 0
```

If the address consists of a single parameter as in the last case, you don't have to add braces (`{}`) around it.

closed 2021-02-10 by tttapa