tttapa/Control-Surface #57: Arduino Esplora Example

tarTG, 2019-11-12:
I tried to use your library with the Arduino Esplora board and got it finally working.

Are you interested to add my code to your examples? Just asking because Arduino does not support the Esplora board any more.
All sensors are handled as potentiometer. Except for the temperature sensor, which crashed my board, and the piezo buzzer.

For the piezo buzzer, it would be nice to have an easy input access based on this [library](https://github.com/bhagman/Tone) if this inside to goals of your work.

```
/**
 * Example for accessing sensors and actuators of the Arduino Esplora board with the 
 * Control Surface library
 * 
 * Currently excluded are the temperature sensor (crashed the Board for me) and the 
 * piezo buzzer.
 * See https://www.arduino.cc/en/Reference/EsploraLibrary for accessing them separately
 * 
 * @boards  Arduino Esplora
 *
 *  Written by tarTG, 2019-12-11  
 *  https://github.com/tttapa/Control-Surface
 */

// Include the library
#include <Esplora.h>
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

//Multiplexer interface
CD74HC4067 mux = {
  A4,       // Analog input pin
  {A0, A1, A2, A3} // Address pins S0, S1, S2
};

//Accelerlometer pins
const byte ACCEL_X_PIN = A5;
const byte ACCEL_Y_PIN = A11;
const byte ACCEL_Z_PIN = A6;

//LED pins
const byte RED_PIN    = 5;
const byte BLUE_PIN   = 9;
const byte GREEN_PIN = 10;

using namespace MIDI_Notes;

// Instantiate a NoteButton object
MIDIButton EsploraSwitches[] = {
    //Switches 
    { mux.pin(CH_SWITCH_1), {note(C, 4), CHANNEL_1} },   // Down
    { mux.pin(CH_SWITCH_2), {note(D, 4), CHANNEL_1} },   // Left
    { mux.pin(CH_SWITCH_3), {note(E, 4), CHANNEL_1} },   // Up
    { mux.pin(CH_SWITCH_4), {note(F, 4), CHANNEL_1} },   // Right
    { mux.pin(CH_JOYSTICK_SW), {note(G, 4), CHANNEL_1} } // Joystick switch
};

//Sensors are handled as potentiometer
CCPotentiometer potentiometer[] = {
  { mux.pin(CH_SLIDER),{MIDI_CC::Channel_Volume, CHANNEL_1}},     // Slider
  { mux.pin(CH_JOYSTICK_X),{MIDI_CC::Channel_Volume, CHANNEL_2}}, // Joystick X-Axis
  { mux.pin(CH_JOYSTICK_Y),{MIDI_CC::Channel_Volume, CHANNEL_3}}, // Joystick Y-Axis
  { mux.pin(CH_LIGHT),{MIDI_CC::Channel_Volume, CHANNEL_5}},      // light sensor
  { mux.pin(CH_MIC),{MIDI_CC::Channel_Volume, CHANNEL_6}},        // microphone
  { ACCEL_X_PIN,{MIDI_CC::Channel_Volume, CHANNEL_7}},            // accelerometer X-axis
  { ACCEL_Y_PIN,{MIDI_CC::Channel_Volume, CHANNEL_8}},            // accelerometer Y-axis
  { ACCEL_Z_PIN,{MIDI_CC::Channel_Volume, CHANNEL_9}}             // accelerometer Z-axis
};

//example for accessing the LEDs
NoteValueLED leds[] = 
{
  {RED_PIN, note(C, 4)},
  {BLUE_PIN, note(D, 4)},
  {GREEN_PIN, note(E, 4)} 
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface 
} 
```

tttapa, 2019-11-13:
Hi, thanks for the example!

I can add it to the "boards" examples if you'd like. I don't own an Esplora myself, so I can't really test it, though.

What exactly would you like the buzzer to do? Should it just play MIDI notes like a MIDI synth?

tarTG, 2019-11-13:
> I can add it to the "boards" examples if you'd like

Feel free to add it.

> Should it just play MIDI notes like a MIDI synth?

Yes. I don`t need this soon. It's just an Idea i came up with while playing around with this board.

tttapa, 2019-11-13:
To play notes, you could use something like this:

```cpp
#include <Control_Surface.h> // Include the Control Surface library

USBMIDI_Interface midi;

// Return true to indicate that handling is done, and Control_Surface shouldn't
// handle it anymore. If you want Control_Surface to handle it as well, return
// false.
bool channelMsgCallback(ChannelMessage cm) {
  ChannelMessageMatcher matcher = cm;
  // Filter the channels if necessary
  if (Channel(matcher.channel) == CHANNEL_1) {
    // Extract note data
    uint8_t note = cm.data1;
    uint8_t velocity = cm.data2;

    if (matcher.type == NOTE_ON) {

      // Play tone here

      return true;
    } else if (matcher.type == NOTE_OFF) {

      // Stop playing tone here

      return true;
    }
  }
  return false;
}

void setup() {
  Serial.begin(115200);
  Control_Surface.begin();
  Control_Surface.setMIDIInputCallbacks(channelMsgCallback, nullptr, nullptr);
}

void loop() {
  Control_Surface.loop();
}
```

closed 2021-02-10 by tttapa