tttapa/Control-Surface #59: STM32 as midi controller

kentforth, 2019-11-18:
Hello Pieter!
I plan to buy stm32 board from aliexpress to build midi controller.
I use Arduino Leonardo for building midi controller. It is simple to turn Arduino to midi controller with Leonardo. 
**What about STM32 board?**
**Should I use any additiional tools/software to use STM32 board as midi controller with your library?**

tttapa, 2019-11-18:
STM32 is not supported by this library.

If you can find a good USB MIDI library for STM32, you can easily add support, though.  
Just implement the `read` and `writeUSBPacket` methods of the `USBMIDI_Interface` class:

<https://github.com/tttapa/Control-Surface/blob/59438364dbb4560f5da37d1b79ec30a24b3f7ed2/src/MIDI_Interfaces/USBMIDI_Interface.hpp#L178-L190>

<https://github.com/tttapa/Control-Surface/blob/59438364dbb4560f5da37d1b79ec30a24b3f7ed2/src/MIDI_Interfaces/USBMIDI_Interface.hpp#L90-L99>

kentforth, 2019-11-18:
Like this one?

tttapa, 2019-11-18:
No, this is not compatible with the STM32 Arduino core, as far as I can tell.

closed 2019-11-28 by kentforth