tttapa/Control-Surface #60: Pitch bend rotary encoder

rogerarends, 2019-11-20:
Hi, could you help me with coding to use an encoder instead of a pot or fader to send pitch bend as a volume control. Thanks in advance.

tttapa, 2019-11-20:
I'd probably use something like this:

```cpp
#include <Control_Surface.h> // Include the Control Surface library
#include <Encoder.h>

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

class PitchBendEncoder : public MIDIOutputElement {
  public:
    PitchBendEncoder(EncoderPinList pins, MIDICNChannel channel)
        : encoder{pins.A, pins.B}, channel{channel} {}
        
    void begin() final override {}

    constexpr static uint8_t RESOLUTION = 6; // bits

    void update() final override {
      noInterrupts();
      long newPosition = encoder.read();
      newPosition = constrain(newPosition, 0, (1 << (RESOLUTION + 2)) - 1);
      encoder.write(newPosition);
      interrupts();
      newPosition >>= 2;
      if (newPosition != previousPosition) {
        PitchBendSender<RESOLUTION>::send(newPosition, channel);
        previousPosition = newPosition;
      }
    }
    
  private:
    Encoder encoder;
    MIDICNChannel channel;
    long previousPosition = -1;
} myEnc = {{0, 1}, CHANNEL_9};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

Basically, to add your own custom elements, you inherit from the `MIDIOutputElement` class to register it with the library. Then you override the `begin` and `update` methods.  
In this case, there's nothing to initialize, so `begin` is empty.  
In the `update` method, you just read the encoder position, constrain it between zero and the maximum value, and then send it if it was updated.  
The position is divided by 4 before sending, because most encoders send out 4 pulses for each physical "click".

rogerarends, 2019-11-20:
Dude! You are amazing…works perfectly, been trying to “build” a custom controller for almost 5 years. Gave up on getting it to work the way I want. Stumbled across your library, found an old Leonardo board, tried it and everything just works. I have no programming skills but most of your library is easy to understand.

Once again, thanks a million.

Roger

closed 2019-11-22 by tttapa