tttapa/Control-Surface #61: how to use two or three or more than Multiplexers 4067 to expand the I/O?

yuanman1988, 2019-11-20:
Hello Pieter!
I have a question about Multiplexers 4067.
I read your example,and success for one Multiplexers 4067 . But I don't know how to use two or three or more than Multiplexers 4067 to expand the I/O. Can you give me some help?
Thank you.

[image]

shikshik, 2019-11-20:
You need to connect each mux to it's EN pin on your Arduino.

`mux1 = {
2,
{3, 4, 5, 6},
7
};`

`mux1 = {
2,
{3, 4, 5, 6},
8
};`

tttapa, 2019-11-20:
What exactly would you like help with?  
The code looks fine, both muxes share the same address lines, and each one has its own input pin.  

In the rest of your code, you can just use the pins of both multiplexers, for example:

```cpp
NoteButton notebuttons[] = {
  {mux1.pin(0), 0x3C}, // first pin of first mux
  {mux2.pin(0), 0x3D}, // first pin of second mux
};
```

yuanman1988, 2019-11-21:
Oh no！
I forgot to delete or modify  the “loop2” in your example,
It led to error...
I'm an idiot.
Now,it runs good.
Anyway, thanks for your reply.And thanks  for your library.It's so cool.

yuanman1988, 2019-11-21:
[image]
[image]
[image]
cool (๑•̀ㅂ•́)و✧

yuanman1988, 2019-11-21:
I have another question.

[image]

What should I do same as “usbMIDI.sendNoteOn(3C, velocity, Channel)” .

tttapa, 2019-11-21:
The signature of the NoteButton constructor is as follows: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d1/d9f/classNoteButton.html#a42d82a21f570f194515c9a7c594c5601>

For example:
```cpp
const uint8_t velocity = 0x7F;
NoteButton buttons[] = {
  {mux1.pin(0), {0x3C, CHANNEL_6}, velocity},
  {mux1.pin(0), {0x3C, CHANNEL_7}, velocity},
};
```
You can also set the velocity afterwards:
```cpp
buttons[0].setVelocity(0x40);
```

yuanman1988, 2019-11-22:
Oh, I see.I need more study.
 I'll try to do it.
Thank you for your help.

closed 2019-11-22 by yuanman1988