tttapa/Control-Surface #62: NoteButton Class is Reversed - NOT WORKING

milkfeet, 2019-11-21:
Hello,
When trying to include a NoteButton in my code (copied the suggested code directly from the guide), I'm encountering an issue where the MIDI input is reversed; when I press the button for the first time in a new session, it will play the note, but if I release the button, the note will continue to play, and will only turn off when I press the button again– essentially the opposite of what I want it to do (playing a note after receiving the "Note OFF" command, and stopping the note when it receives a "Note ON" command). Not sure if I'm doing something wrong, but I have tried with different types of buttons and in different pins. It seems like I can not get it to function properly no matter what I do (which makes me think it might be an error in the definition of the NoteButton?). Please let me know if you have any suggestions on how I might be able to fix this problem.

tttapa, 2019-11-21:
I'm unable to reproduce this problem. 

Please post your exact code and wiring, as well as more details about the Arduino board you're using and the software you're trying it with.  

One pin of the push button should be connected to ground, the other to the right Arduino digital pin.  
If you have a normally closed (NC) switch, you can invert the button (`button.invert()`).

samlochner, 2019-11-22:
Same problem, but button.invert() doesn't seem to work for me. I think I must not be using it correctly. I get the error 'button does not name a type'. What is the correct code for this?

Thanks Sam

my code:
```
NoteButton button = {7, {72, CHANNEL_1},127};
button.invert();
```

tttapa, 2019-11-22:
Please post the full sketch.

Statements like `button.invert();` go in a function like `setup`.

Did you try the NoteButton example without anything else?

samlochner, 2019-11-22:
I didn't have it in a function, that was the problem... still quite new to Arduino's....
Thanks for the quick response!

tttapa, 2019-11-22:
Does the button work now?

samlochner, 2019-11-22:
Yup!

tttapa, 2019-11-22:
Glad to hear!

closed 2019-12-02 by tttapa