tttapa/Control-Surface #63: send ppqn via usb

2kohm, 2019-11-21:
hi,
is it possible to send 

- // Clock start = 0xFA
- // Clock stop = 0xFC
- // Clock pulse = 0xF8

via USB with this library ?

tttapa, 2019-11-21:
Yes, I just added it in 461a67fee2065d7669cdcdb5aa5f75e430070691.

```cpp
#include <Control_Surface.h>

// Instantiate the MIDI over USB interface
USBMIDI_Interface midi;

// Push button connected between pin 2 and ground.
// MIDI Clock message is sent when pressed.
Button pushbutton = {2};  

void setup() {
  pushbutton.begin();  // enables internal pull-up
  midi.begin();
}

void loop() {
  if (pushbutton.update() == Button::Falling)
    midi.send(0xF8);
  midi.update();
}
```

2kohm, 2019-11-22:
awesome, thanks for the fast replay and adding this!!!!

closed 2019-11-22 by 2kohm

2kohm, 2019-11-23:
mh

`Arduino\libraries\Control-Surface-master\src/MIDI_Inputs/MCU/VU.hpp:90:16: warning: missing initializer for member 'Array<unsigned char, 1u>::data' [-Wmissing-field-initializers]

         values = {};

                ^`