tttapa/Control-Surface #64: can`t compile

2kohm, 2019-11-23:
`Arduino\libraries\Control-Surface-master\src/MIDI_Inputs/MCU/VU.hpp:90:16: warning: missing initializer for member 'Array<unsigned char, 1u>::data' [-Wmissing-field-initializers]

     values = {};

            ^`

tttapa, 2019-11-23:
This is strange, it looks like a bug in the Due Core. The pointer is const, but the pointee is not.

I'll push a work-around in just a minute.

tttapa, 2019-11-23:
I would think that the correct code should be:

```cpp
#define pgm_read_ptr(addr) (*(void *const *)(addr))
```
(A pointer to a read-only pointer to a mutable object of any type.)

tttapa, 2019-11-23:
I noticed some other compilation problems on Due, I'll fix them as soon as possible. The reason is that they are using a very old compiler.

tttapa, 2019-11-23:
One of them is a bug in GCC that has been fixed in August of 2017, but the Due uses an ancient compiler from 2014.  
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=47226

```
In file included from /home/pieter/Arduino/libraries/Control-Surface/src/MIDI_Constants/Chords/Chords.hpp:4:0,
                 from /home/pieter/Arduino/libraries/Control-Surface/src/MIDI_Outputs/Abstract/MIDIChordButton.hpp:6,
                 from /home/pieter/Arduino/libraries/Control-Surface/src/MIDI_Outputs/NoteChordButton.hpp:3,
                 from /home/pieter/Arduino/libraries/Control-Surface/src/Control_Surface.h:43,
                 from /tmp/arduino_modified_sketch_595653/MCU-OLED-SSD1306-x2.ino:61:
/home/pieter/Arduino/libraries/Control-Surface/src/Helpers/ArrayHelpers.hpp: In lambda function:
/home/pieter/Arduino/libraries/Control-Surface/src/Helpers/ArrayHelpers.hpp:115:47: error: parameter packs not expanded with '...':
     return generateArray<N>([&]() { return T{args...}; });
                                               ^
/home/pieter/Arduino/libraries/Control-Surface/src/Helpers/ArrayHelpers.hpp:115:47: note:         'args'
/home/pieter/Arduino/libraries/Control-Surface/src/Helpers/ArrayHelpers.hpp:115:51: error: expansion pattern 'args' contains no argument packs
     return generateArray<N>([&]() { return T{args...}; });
```

tttapa, 2019-11-23:
I opened an issue in the ArduinoCore-sam repository for the `pgm_read_ptr` bug.

The other errors I'm getting when compiling for Due are because of a defect in the C++11 standard. It has been fixed since (June, 2014), but again, the Due compiler is ancient and doesn't have the fix yet.  
<http://www.open-std.org/jtc1/sc22/wg21/docs/cwg_defects.html#1467>

tttapa, 2019-11-23:
I added a branch with workarounds for Due: https://github.com/tttapa/Control-Surface/tree/due-old-gcc

2kohm, 2019-11-24:
Thanks alot !!
`Linking .pio\build\dueUSB\firmware.elf
Checking size .pio\build\dueUSB\firmware.elf
Building .pio\build\dueUSB\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=         ]   5.1% (used 4976 bytes from 98304 bytes)
PROGRAM: [          ]   2.9% (used 15300 bytes from 524288 bytes)
[SUCCESS] Took 17.10 seconds

commit eee59dbc (2019-11-23): Fix #64
closed 2019-11-28 by tttapa