tttapa/Control-Surface #65: send note on / off on events

2kohm, 2019-11-24:
First, thanks again for the arduino due fix
i am working on a usbmidi send clock.

`void clock(){
 midi.send(0xF8);
++tick;
midi.update();
}`
works perfect so far.

is it possible to send note on and off events like with the MIDIUSB lib 

`void noteOn(byte channel, byte pitch, byte velocity) {
  midiEventPacket_t noteOn = {0x09, 0x90 | channel, pitch, velocity};
  MidiUSB.sendMIDI(noteOn);
}

void noteOff(byte channel, byte pitch, byte velocity) {
  midiEventPacket_t noteOff = {0x08, 0x80 | channel, pitch, velocity};
  MidiUSB.sendMIDI(noteOff);
}`

with your lib ? since it is bounded to button presses in your lib

` if (something) {
noteOn(0,60,127);
`

tttapa, 2019-11-24:
You can use the `sendNoteOn` method of the MIDI_Interface class: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d9/d73/classMIDI__Interface.html#a80c4843f88185a213ebe20a26c0fd579

The channel and cable number are optional. If you only specify the note number, you don't need braces.

For example:
```cpp
USBMIDI_Interface midi;

[...]

midi.sendNoteOn({0x3C, CHANNEL_3, 10}, 0x7F); // Note 0x3C (middle C), Channel 3, Cable 10, Velocity 0x7F
midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);     // Cable 0 (first and default cable)
midi.sendNoteOn(0x3C, 0x7F);                  // Channel 1, Cable 0
```

tttapa, 2019-11-24:
But you only need that if you manually have to send the note events.

If you want a button or switch that sends MIDI note events when pressed, you can use the [NoteButton](https://tttapa.github.io/Control-Surface/Doc/Doxygen/de/dcc/NoteButton_8ino-example.html) class.

2kohm, 2019-11-24:
thanks again for your fast replay! clock generation on the due works perfect with 96 ppqn resolution!

another question :D 

how would i do 
`  midiEventPacket_t rx;
  do {
    rx = MidiUSB.read();
    if (rx.header != 0) {
      Serial.print("Received: ");
      Serial.print(rx.header, HEX);
      Serial.print("-");
      Serial.print(rx.byte1, HEX);
      Serial.print("-");
      Serial.print(rx.byte2, HEX);
      Serial.print("-");
      Serial.println(rx.byte3, HEX);
    }
  } while (rx.header != 0);`
with your lib ? i wan't to read incoming clock signals

tttapa, 2019-11-24:
Glad to hear!

You can add custom callbacks to the MIDI inputs: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d0/d32/MIDI-Input-Callback_8ino-example.html

2kohm, 2019-11-24:
ok, this works fine, `Real-Time message: f8 on cable 0`
but how can grab this clock event ?

```
if (serial.read == 0xf8) {
};
```

tttapa, 2019-11-24:
You can add your own code in the `realTimeMessageCallback` function, and then use [`rt.message`](https://tttapa.github.io/Control-Surface/Doc/Doxygen/d6/d99/structRealTimeMessage.html).

```cpp
bool realTimeMessageCallback(RealTimeMessage rt) {
  if (rt.message == 0xF8) {
    // Do what you need to do if a clock event is received
  }
  return true;
}
```

closed 2019-12-02 by tttapa