tttapa/Control-Surface #66: Variable CC message on potentiometers

Rolbista, 2019-11-27:
Hello

I am building a MIDI controller using your library - the goal is to have two modwheels with assignable CC. To achieve this I was thinking about using the `CCPotentiometer` class and updating its property `address` with a value of 0-119 selected on a rotary encoder - however I can't see a straightforward way to do this. What would be your suggestion to achieve free CC value change to a potentiometer? A bank with 120 values? 

The library I'm looking at is at commit `19d2efc7037c2e176feca44750a12594c76f466f`.

tttapa, 2019-11-27:
I added an example: <https://tttapa.github.io/Control-Surface/Doc/Doxygen/d1/da6/Encoder-Selector-Bank_8ino-example.html>

Is that what you had in mind?

Rolbista, 2019-11-29:
At first glance looks almost exactly like what I need, I'll upload this to my board to test. Can I use one encoder selector for two different potentiometers by introducing some logic or do they need a selector for each? So CC can be changed independently for each wheel?

As I understand, `bank.getSelection();` can be used to for example display current CC on a segment display?

tttapa, 2019-11-29:
I think it would be easiest to handle the rotary encoder yourself if you need it to control two different settings.

You can create two banks, one for each potentiometer, and then use the [Bank::select](https://tttapa.github.io/Control-Surface/Doc/Doxygen/db/dbd/classBank.html#ab4b91b7f4363e5e497b34e17a303bdbb) method.

To display the bank setting, you can use [MAX7219::display](https://tttapa.github.io/Control-Surface/Doc/Doxygen/d4/d26/classMAX7219SevenSegmentDisplay.html#a03c3e91c6342104c691f89ba17b99083).

Rolbista, 2019-11-30:
Alright, looking good, thank you very much!

closed 2019-11-30 by Rolbista