tttapa/Control-Surface #67: Wireless Midi Controller with bluetooth module

kentforth, 2019-11-28:
Hello Pieter!
I plan to make wireless midi controller using Arduino Leonardo.

**Do you have implementation of bluetooth module?**

**Can you give me an example  of code how to use two buttons with bluetooth?**

**What type of bluetooth module should I use? How to connect it to Arduino?**

tttapa, 2019-11-28:
Right now, only the ESP32 is supported: https://tttapa.github.io/Control-Surface/Doc/Doxygen/d0/d89/classBluetoothMIDI__Interface.html

kentforth, 2019-11-28:
Can you give me an example of code for ESP32 board?

tttapa, 2019-11-28:
It's exactly the same, you just use `BluetoothMIDI_Interface` instead of `USBMIDI_Interface`.

kentforth, 2019-11-28:
There are different ESP32 boards. What microcontroller should true ESP32 board have to make it as midi controller with bluetooth?

I found board like this, is it good for my purpose?

https://ru.aliexpress.com/item/32996463686.html?spm=a2g0o.productlist.0.0.24d44037gzqT41&algo_pvid=18dd59f9-f94e-4fab-aa45-a14edb615fa6&algo_expid=18dd59f9-f94e-4fab-aa45-a14edb615fa6-0&btsid=3e996789-4dac-45cd-9f32-56eea6c54cad&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_53

tttapa, 2019-11-28:
ESP32 _is_ the microcontroller.  
I'm not up to date with the different models, I use a Sparkfun ESP32 Thing because it was the first one that was available locally where I live, but that was a long time ago. 
If it has an ESP32 chip, a USB interface and enough pins, it should be alright.

kentforth, 2019-11-28:
Thank you Pieter!

closed 2019-11-28 by kentforth