tttapa/Control-Surface #69: Issue using LEDs for Control Change values without multiplexor

finntgeorge, 2019-12-04:
Hey we have a controller set up with a few knobs and a button and we are running into some issues triggering the 4 LEDs plugged into the digital pins all at once with one of the potentiometers. Here is our code so far, any suggestions are appreciated. 

`#include <Control_Surface.h>
#include <MIDIUSB.h>
// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

// Instantiate objects
CCButton button = {12, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5} };
CCPotentiometer fader = { A0, {MIDI_CC::Channel_Volume, CHANNEL_1} };
CCPotentiometer knob1 = { A1, {MIDI_CC::Channel_Volume, CHANNEL_2} };
CCPotentiometer knob2 = { A2, {MIDI_CC::Channel_Volume, CHANNEL_3} };
CCPotentiometer knob3 = { A3, {MIDI_CC::Channel_Volume, CHANNEL_4} };
CCValueLED leds[] = {
  {pin_t(3), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}}, // LED pin, address (note number, channel, cable)
  {pin_t(2), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}},
  {pin_t(4),{MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}},
  {pin_t(5),{MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}},
};

// Initialize the Control Surface
void setup() {
  Control_Surface.begin();
  fader.invert();
 }

// Update the Control Surface
void loop() {
  Control_Surface.loop();
 }`

tttapa, 2019-12-05:
Hi,
Could you be more specific?

I just tested the CCValueLED class, and everything seems to work as expected.

It should be noted that the LEDs do not read the input from the button on the same address, they read the input from the MIDI interface. So if you want the LEDs to do anything, you have to send it to the Arduino over MIDI. (Or you have to program turn on/off the LED manually.)

Second, having multiple LEDs with the same address is not supported, only the first will match the incoming MIDI message.

tttapa, 2019-12-05:
If you want to turn on the LED when the button is pressed, you could use something like this:

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

CCButton button = {12, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}};

constexpr pin_t ledPin = 13;

void setup() {
  Control_Surface.begin();
  pinMode(ledPin, OUTPUT);
}

void loop() {
  Control_Surface.loop();
  if (button.getButtonState() == Button::Rising)
    digitalWrite(ledPin, LOW);
  else if(button.getButtonState() == Button::Falling)
    digitalWrite(ledPin, HIGH);
}
```

closed 2021-02-10 by tttapa