tttapa/Control-Surface #71: How to use encoders with multiplexers?

kentforth, 2019-12-08:
Hello Pieter!
Is it possible to add implementation of multiplexers to encoders?

tttapa, 2019-12-08:
No, this is not possible. Rotary encoders require interrupts to work correctly. If the loop is fast enough, you might be able to use polling, but I don't have time to add it to the library.

closed 2019-12-08 by kentforth

tttapa, 2019-12-08:
You can relatively easily write the polling code yourself, and then use the [`RelativeCCSender`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dd0/classRelativeCCSender.html) class to send the MIDI messages.

You can use this for reference:  
https://github.com/tttapa/Control-Surface/blob/0e1b921f1cc329bcb175483174212e57e00903f0/src/MIDI_Outputs/Abstract/MIDIRotaryEncoder.hpp#L46-L53