tttapa/Control-Surface #72: send notes on 2 channels at the same time

2kohm, 2019-12-08:
Hello tttapa,

is it not possible to send 2 notes on 2 channels at the same time ?
```
	 if (stepts == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);
	 }
	 if(stepts == 96) {
		 midi.sendNoteOff({0x3C, CHANNEL_3}, 0x00);
	 }

   	 if (stepts2 == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_4}, 0x7F);

	 }
	 if(stepts2 == 96) {
		 midi.sendNoteOff({0x3C, CHANNEL_4}, 0x00);

	 }
```
it only sends the note on channel 3
if i change it to
```
	 if (stepts == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);
	 }
	 if(stepts == 96) {
		 midi.sendNoteOff({0x3C, CHANNEL_3}, 0x00);
	 }

   	 if (stepts2 == 2) {
		 midi.sendNoteOn({0x3C, CHANNEL_4}, 0x7F);

	 }
	 if(stepts2 == 97) {
		 midi.sendNoteOff({0x3C, CHANNEL_4}, 0x00);

	 }
```
it works fine. any ideas ? for the counters i use **<DueTimer.h>**

2kohm, 2019-12-08:
ok, this have something to do with DueTimer.h

tttapa, 2019-12-08:
Sending two notes on 2 channels should just work. Could you post your complete sketch?

2kohm, 2019-12-08:
```
#include <Arduino.h>
#include <MIDIUSB.h>
#include <Control_Surface.h>
#include <DueTimer.h>
#include <RotaryEncoder.h>
USBMIDI_Interface midi;

const int midiClock = 0xf8;    // midic clock pulse byte
const int startByte = 0xfa;    // midi clock start byte
const int stopByte = 0xFC;     // midi clock stop byte

// rotary encoder pins
const int SW = 46;            
const int CLK = 48;
const int DT = 50;
int ticks = 0;
int stepts = 0;
int stepts2 = 0;

// initializing the encoder object
RotaryEncoder Encoder(DT,CLK);

int encoderValue = 120;
int encoderValueOLD = 120;

// start bpm (beats per minute)
int bpm = 120;        
long interval = 60000000 / (96 * bpm);

bool startclock = true;

void update_TEMPO() {
 interval = 60000000 / (96 * bpm);

 Timer0.start(interval);   // midi clock
 Timer1.start(interval);   // encoder
}

void send_MIDICLOCK() {
 if (startclock) {
	 ticks++;
	 stepts++;
   stepts2++;
	 if(ticks == 0) {
		 midi.send(midiClock);
	 }
	  if(ticks == 4) {
		 midi.send(midiClock);
		 ticks = 0;
	 }

	 if (stepts == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);
	 }
	 if(stepts == 9) {
		 midi.sendNoteOff({0x3C, CHANNEL_3}, 0x00);
	 }

   	 if (stepts2 == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_4}, 0x7F);

	 }
	 if(stepts2 == 96) {
		 midi.sendNoteOff({0x3C, CHANNEL_4}, 0x00);

	 }

	if(stepts == 384) {
		stepts = 0;
	}
  	if(stepts2 == 384) {
		stepts = 0;
	}

 }
}

void midiclock_STARTSTOP() {
 startclock = !startclock;

 if (startclock) {
   stepts = 0;
    stepts2 = 0;
   Timer0.start(interval);
   midi.send(startByte);
 }
 else {
   Timer0.stop();
  midi.send(stopByte);
  ticks = 0;
  stepts = 0;
  stepts2 = 0;
 }
}

void read_ENCODER() {
 
 Encoder.tick();
 encoderValue = Encoder.getPosition();
 
 if (encoderValue != encoderValueOLD) {
   encoderValueOLD = encoderValue;

   if (encoderValue > 300) {
     encoderValue = 300;
   }
   else if (encoderValue < 30) {
     encoderValue = 30;
   }

   bpm = encoderValue;
   Encoder.setPosition(bpm);
   
   SerialUSB.println(bpm);
   update_TEMPO();
 }
}

void setup() {
 pinMode(SW, INPUT);
 attachInterrupt(digitalPinToInterrupt(SW), midiclock_STARTSTOP, RISING);
 Encoder.setPosition(bpm);
 Timer0.attachInterrupt(send_MIDICLOCK);
 Timer1.attachInterrupt(read_ENCODER);
 Timer1.start(interval);
}

void loop() {

}

```

tttapa, 2019-12-08:
Using USB and Serial in interrupt handlers is bad practice, because they require interrupts to work correctly, and interrupts are disabled inside of interrupt handlers.

The handlers should be as short as possible. Just set a flag, and then handle the sending of data over USB in the main loop.

2kohm, 2019-12-08:
thanks for the hint

2kohm, 2019-12-08:
so the only problem is, how would i get the timeintervall in in the main loop ?
when i do this

```
void send_MIDICLOCK() {
 if (startclock) {
	 ticks++;
	 stepts++;
   stepts2++;
	}
void loop {
 if(ticks == 0) {
		 midi.send(midiClock);
	 }
	  if(ticks == 4) {
		 midi.send(midiClock);
		 ticks = 0;
	 }

	 if (stepts == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);
	 }
	 if(stepts == 9) {
		 midi.sendNoteOff({0x3C, CHANNEL_3}, 0x00);
	 }
}

```
it send´s the note to often
[image]

2kohm, 2019-12-08:
this is what you ment with flag and handle it in main loops ?
```
bool start = false;
void myHandler(){
  start = true;
}

void setup(){
	pinMode(myLed, OUTPUT);

	Timer3.attachInterrupt(myHandler);
	Timer3.start(500000); // Calls every 50ms
}

void loop(){

if(start == true) {
  SerialUSB.println("ture");
  start = false;
}
	
}
```

2kohm, 2019-12-08:
and if i do this, only channel 3 sends notes
```
void loop() {

  if(start == true) {
    	 if(ticks == 0) {
		 midi.send(midiClock);
	 }
	  if(ticks == 4) {
		 midi.send(midiClock);
		 ticks = 0;
	 }

	 if (stepts == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);
	 }
	 if(stepts == 9) {
		 midi.sendNoteOff({0x3C, CHANNEL_3}, 0x00);
	 }

   	 if (stepts == 1) {
		 midi.sendNoteOn({0x3C, CHANNEL_4}, 0x7F);

	 }
	 if(stepts == 96) {
		 midi.sendNoteOff({0x3C, CHANNEL_4}, 0x00);

	 }

	if(stepts == 384) {
		stepts = 0;
	}
  start = false;
  }
```

tttapa, 2019-12-08:
That looks alright. Not sure why channel 4 doesn't work though.

Could you try a simple example, something like this:

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

void setup() {
  midi.begin();
}

void loop() {
  midi.sendNoteOn({0x3C, CHANNEL_3}, 0x7F);
  midi.sendNoteOn({0x3C, CHANNEL_4}, 0x7F);
  midi.update();
  delay(500);
  midi.sendNoteOff({0x3C, CHANNEL_3}, 0x7F);
  midi.sendNoteOff({0x3C, CHANNEL_4}, 0x7F);
  midi.update();
  delay(500);
}
```

I don't own an Arduino Due, so I can't test it myself.

2kohm, 2019-12-08:
sends only on channel 3

tttapa, 2019-12-08:
Does using just channel 4 work?

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

void setup() {
  midi.begin();
}

void loop() {
  midi.sendNoteOn({0x3C, CHANNEL_4}, 0x7F);
  midi.update();
  delay(500);
  midi.sendNoteOff({0x3C, CHANNEL_4}, 0x7F);
  midi.update();
  delay(500);
}
```
What software are you using at the other side?

2kohm, 2019-12-08:
yes

2kohm, 2019-12-08:
this works random, but strange behavior, 
```
#include <Arduino.h>
#include <MIDIUSB.h>
#include <Control_Surface.h>

USBMIDI_Interface midi;

void setup() {
  midi.begin();
}

void loop() {
  midi.sendNoteOn({0x3C, CHANNEL_1}, 0x7F);
  midi.sendNoteOn({0x3C, CHANNEL_2}, 0x7F);
  midi.update();
  delay(500);
  midi.sendNoteOff({0x3C, CHANNEL_1}, 0x7F);
  midi.sendNoteOff({0x3C, CHANNEL_2}, 0x7F);
  midi.update();
  delay(500);
}
```
[image]

2kohm, 2019-12-08:
Ableton Live 10

2kohm, 2019-12-08:
[image]

tttapa, 2019-12-08:
Very strange.

Could you try the same thing with just the MIDIUSB library:
```cpp
#include <MIDIUSB.h>

void setup() {}

void loop() {
  MidiUSB.sendMIDI({0x09, 0x92, 0x3C, 0x7F});
  MidiUSB.flush();
  MidiUSB.sendMIDI({0x09, 0x93, 0x3C, 0x7F});
  MidiUSB.flush();
  delay(500);
  MidiUSB.sendMIDI({0x08, 0x82, 0x3C, 0x7F});
  MidiUSB.flush();
  MidiUSB.sendMIDI({0x08, 0x83, 0x3C, 0x7F});
  MidiUSB.flush();
  delay(500);
}
```

2kohm, 2019-12-08:
[image]

same :/

2kohm, 2019-12-08:
this works
#include <Arduino.h>
#include <MIDIUSB.h>

```
void setup() {}

void loop() {
  MidiUSB.sendMIDI({0x09, 0x92, 0x3C, 0x7F});
  MidiUSB.flush();
  delayMicroseconds(50);
  MidiUSB.sendMIDI({0x09, 0x93, 0x3C, 0x7F});
  MidiUSB.flush();
  delay(500);
  MidiUSB.sendMIDI({0x08, 0x82, 0x3C, 0x7F});
  MidiUSB.flush();
  delayMicroseconds(50);
  MidiUSB.sendMIDI({0x08, 0x83, 0x3C, 0x7F});
  MidiUSB.flush();
  delay(500);
}
```

[image]

tttapa, 2019-12-08:
Well, this does mean that it's not a bug in the Control Surface library, but in the MIDIUSB library. Sadly, this doesn't solve your problem, of course.

Could you try flushing after sending the two notes?

```cpp
#include <MIDIUSB.h>

void setup() {}

void loop() {
  MidiUSB.sendMIDI({0x09, 0x92, 0x3C, 0x7F});
  MidiUSB.sendMIDI({0x09, 0x93, 0x3C, 0x7F});
  MidiUSB.flush();
  delay(500);
  MidiUSB.sendMIDI({0x08, 0x82, 0x3C, 0x7F});
  MidiUSB.sendMIDI({0x08, 0x83, 0x3C, 0x7F});
  MidiUSB.flush();
  delay(500);
}
```

2kohm, 2019-12-08:
[image]

tttapa, 2019-12-08:
This seems to be a known problem with the MIDIUSB library:

https://github.com/arduino-libraries/MIDIUSB/issues/44

It's a shame that the Due is so poorly supported ...

2kohm, 2019-12-08:
sadly

commit 257637a4 (2019-12-11): Add delay on MIDIUSB flush for Arduino Due
Fixes #72
closed 2019-12-08 by 2kohm