tttapa/Control-Surface #73: Oled with Mega 2560

rogerarends, 2019-12-09:
Hi.
Is it possible to update your MCU oled sketch to include Arduino Mega support. I only need track name and pan for the first channel. Or fix the sketch for MCU LED VU meter. It won't compile. Maybe an example without using the shiftfegister. I'm using Presonus Studio one 4 as my daw. I just purchased a Teensy 4 but unfortunately didn't check support so it's useless for now.
The controller I built is similar to the Beringer X touch one. 

Thanks in advance

rogerarends, 2019-12-09:
Or I was thinking I could use hairless midi interface with the teensy 4 using your library. That would solve the problem for now until usbmidi is implemented.

tttapa, 2019-12-09:
1c54ba6d4910150bea0303583673611f29def824

I edited the MCU OLED sketch for the Mega, the only thing that had to be changed were the pin numbers and configuration.

I also added a VULEDs example.

rogerarends, 2019-12-09:
Thanks, I'll test it tonight. Also regarding the teensy 4. Can I use it with hairless or should I wait till there's usbmidi implemented. On the forums they say it should be this month

1c54ba6<https://github.com/tttapa/Control-Surface/commit/1c54ba6d4910150bea0303583673611f29def824>

tttapa, 2019-12-09:
You should be able to use hairless without problems.

closed 2019-12-13 by rogerarends

Alexandr010360, 2024-01-18:
Hello! I can't find Full [MCU-OLED-SSD1306-MEGA.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d86/MCU-OLED-SSD1306-MEGA_8ino-example.html ) on the website! Please send it to me by e-mail.